ladislas/Bare-Arduino-Project #12: Arduino-Makefile submodule unable to checkout

cgel, 2015-04-14:
i follow the instructions in INSTALL.md and when using the command:
git submodule update --init --recursive
i get:
fatal: reference is not a tree: d1ea8699a49962f7c58beeaa25878755e0910195
Unable to checkout 'd1ea8699a49962f7c58beeaa25878755e0910195' in submodule path 'Arduino-Makefile'

ladislas, 2015-04-14:
I get the same error, will fix this asap :)

ladislas, 2015-04-18:
@cgel should work now.

tell me if everything is OK :)

closed 2015-04-18 by ladislas

cgel, 2015-04-18:
the submodule update works great :)
But when i tried to compile BarProject the compiler does not have the path to my libs/ 
The USER_LIB_PATH is well defined.

ladislas, 2015-04-18:
can you open a new issue and copy past the output of `make` please :)

PR #13 (2015-04-18): Add Arduino IDE 1.6.3 compatibility and update automatic lib detection
Fixes #12