ladislas/Bare-Arduino-Project #14: Transitive library dependencies

bssstudio, 2015-04-18:
The sample project wont compile.

I am compiling `src/BarProject` that uses `BarLib`

It does not find the FooLib library that is a dependency of `BarLib`
```
/tmp/path/to/New_Arduino_Project/lib/BarLib/BarLib.h:11:20: fatal error: FooLib.h: No such file or directory
 #include "FooLib.h"
```

However if I specify both libraries in the make file:
```
ARDUINO_LIBS = Wire BarLib FooLib 
```
it compiles.

Whole compile log:
```
$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX
- [USER]               ARDUINO_DIR = /usr/share/arduino
- [USER]               ARDMK_DIR = /tmp/path/to/New_Arduino_Project/Arduino-Makefile
- [AUTODETECTED]       ARDUINO_VERSION = 163
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            VENDOR = arduino
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/user/.arduino15/preferences.txt
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/user/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /tmp/path/to/New_Arduino_Project/lib
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_SUB = atmega2560
- [USER]               BOARD_TAG = mega
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = mega (from build.variant)
- [USER]               OBJDIR = /tmp/path/to/New_Arduino_Project/bin/BarProject/mega
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
- [USER]               MONITOR_BAUDRATE = 115200
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [USER]               CFLAGS_STD = -std=gnu11
- [USER]               CXXFLAGS_STD = -std=gnu++11
- [COMPUTED]           DEVICE_PATH = /dev/tty.usbmodem* (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      SYSTEM_LIBS =
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.3.4
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------
mkdir -p /tmp/path/to/New_Arduino_Project/bin/BarProject/mega
/usr/bin/avr-g++ -MMD -c -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=163 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/mega    -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/utility   -I/tmp/path/to/New_Arduino_Project/lib/BarLib -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fdiagnostics-color -fno-exceptions -std=gnu++11 BarProject.cpp -o /tmp/path/to/New_Arduino_Project/bin/BarProject/mega/BarProject.o
In file included from BarProject.cpp:3:0:
/tmp/path/to/New_Arduino_Project/lib/BarLib/BarLib.h:11:20: fatal error: FooLib.h: No such file or directory
 #include "FooLib.h"
                    ^
compilation terminated.
/tmp/path/to/New_Arduino_Project/Arduino-Makefile/Arduino.mk:1238: recipe for target '/tmp/path/to/New_Arduino_Project/bin/BarProject/mega/BarProject.o' failed
m
```

ladislas, 2015-04-18:
you should not need to define  `ARDUINO_LIBS`,  they should be automatically added.

the `/tmp/path/to/New_Arduino_Project` seems strange, can you `pwd` inside the root of the project and copy/paste the output?

bssstudio, 2015-04-18:
This is the actual root of the project. I was testing it and cloned into `/tmp/path/to/New_Arduino_Project/`. I followed the install guide and I regularly use /tmp for testing stuff :)

ladislas, 2015-04-18:
can you copy/paste the content of the makefile
I just tested like you do and it works great for me.

bssstudio, 2015-04-18:
Sure :)

```
### DISCLAIMER
### This is an example Makefile and it MUST be configured to suit your needs.
### For detailled explanations about all the avalaible options,
### please refer to https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md

### PROJECT_DIR
### This is the path to where you have created/cloned your project
PROJECT_DIR       = /tmp/path/to/New_Arduino_Project

### AVR_GCC_VERSION
### Check if the version is equal or higher than 4.9
AVR_GCC_VERSION  := $(shell expr `avr-gcc -dumpversion | cut -f1` \>= 4.9)

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(PROJECT_DIR)/Arduino-Makefile

### ARDUINO_DIR
### Path to the Arduino application and ressources directory.
ARDUINO_DIR       = /usr/share/arduino

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
USER_LIB_PATH     :=  $(realpath $(PROJECT_DIR)/lib)

### BOARD_TAG & BOARD_SUB
### It must be set to the board you are currently using. (i.e uno, mega, etc.)
### For the Arduino Uno, only BOARD_TAG is mandatory and BOARD_SUB can be equal to anything
### For the Arduino Mega2560, BOARD_SUB is also needed
BOARD_TAG         = mega
BOARD_SUB         = atmega2560

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 115200

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
AVR_TOOLS_DIR     = /usr

### AVRDDUDE
### Path to avrdude directory.
AVRDDUDE          = /usr/bin/avrdude

### CFLAGS_STD
CFLAGS_STD        = -std=gnu11

### CXXFLAGS_STD
CXXFLAGS_STD      = -std=gnu++11

### CPPFLAGS
### Flags you might want to set for debugging purpose. Comment to stop.
CXXFLAGS         = -pedantic -Wall -Wextra

### If avr-gcc -v is higher than 4.9, activate coloring of the output
ifeq "$(AVR_GCC_VERSION)" "1"
    CXXFLAGS += -fdiagnostics-color
endif

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT      = /dev/tty.usbmodem*

### don't touch this
CURRENT_DIR       = $(shell basename $(CURDIR))

### OBJDIR
### This is were you put the binaries you just compile using 'make'
CURRENT_DIR       = $(shell basename $(CURDIR))
OBJDIR            = $(PROJECT_DIR)/bin/$(CURRENT_DIR)/$(BOARD_TAG)

### path to Arduino.mk, inside the ARDMK_DIR, don't touch.
include $(ARDMK_DIR)/Arduino.mk
```

ladislas, 2015-04-18:
can you `cd` inside `Arduino-Makefile` and tell me what branch you are using?

ladislas, 2015-04-18:
it should be `auto-lib` and **not** `master`

bssstudio, 2015-04-18:
HEAD detached at 807e387 which is 
```
commit 807e38753bbf2333d5b030c08247efb7b0aceeb6
Merge: eba20c7 02382b2
Author: Ladislas de Toldi <ladislas@leka.io>
Date:   Sun Apr 5 20:28:37 2015 +0200

    add automatic lib detection
```
Gitk says it's auto-lib branch.

ladislas, 2015-04-18:
that's strange, I don't understand why it doesn't work... :(

could you try cloning the project again?

bssstudio, 2015-04-18:
I tried... no success :(
Here are the steps I took (maybe i'm doing something utterly wrong :))
```
cd /tmp
git clone https://github.com/WeAreLeka/Bare-Arduino-Project path/to/New_Arduino_Project
cd path/to/New_Arduino_Project/
git submodule update --init --recursive
cd src/BarProject/
cp ../../Makefile-Linux.mk Makefile
vim Makefile ## edit project path to /tmp/path/to/New_Arduino_Project
make
```

ladislas, 2015-04-19:
I think I found the issue!

I just followed you exact same steps on my linux vm and it works. In the output I have something like this:

```
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      SKETCH_LIBS =
- [USER]                 BarLib
-
-                      SKETCH_LIBS_DEPS =
- [USER]                 FooLib
-
-                      SYSTEM_LIBS =
- [SYSTEM]               Wire
-
```

Which is not present in your output. Can you list the content of `/path/to/your-project/Arduino-Makefile/bin`? Also, do you have `python` installed?

ladislas, 2015-04-23:
@bssstudio have you had any chance of trying what I said?

ladislas, 2015-06-22:
@bssstudio I'm closing the issue now, feel free to reopen it if the issue is still there.

alexozer, 2015-07-09:
I have the same issue on my Linux installation. I am using the auto-lib branch of Arduino-Makefile and yet it still fails to find the "FooLib.h" header included in BarProject.cpp.

Here's my Arduino-Makefile/bin directory contents:
```
ard-reset-arduino  lib-detection
```
and so the lib-detection script exists.

On a side note, shouldn't BarProject.cpp include FooLib.h like this:
```
#include <FooLib.h>
```
instead of:
```
#include "FooLib.h"
```
since headers in lib/ correspond to system-wide (not project-local) headers? Either way though, it doesn't work.

ladislas, 2015-07-10:
@alexozer could you copy/paste your makefile please?

I just tried again from a brand new clone, both on my computer and on my remote server and I don't have any issues.

What version of linux are you using? Are you able to compile the `FooProject` ?

regarding the use of `" "` insteand of `< >`: `<Arduino.h>` is a system wide library but `"FooLib.h"` is a project library as Bare-Arduino-Project is a per project template.

alexozer, 2015-07-10:
Sure, here's my Makefile:
```
### DISCLAIMER
### This is an example Makefile and it MUST be configured to suit your needs.
### For detailled explanations about all the avalaible options,
### please refer to https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md

### PROJECT_DIR
### This is the path to where you have created/cloned your project
PROJECT_DIR       = /home/alex/build/Bare-Arduino-Project

### AVR_GCC_VERSION
### Check if the version is equal or higher than 4.9
AVR_GCC_VERSION  := $(shell expr `avr-gcc -dumpversion | cut -f1` \>= 4.9)

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(PROJECT_DIR)/Arduino-Makefile

### ARDUINO_DIR
### Path to the Arduino application and ressources directory.
ARDUINO_DIR       = /usr/share/arduino

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
USER_LIB_PATH     :=  $(realpath $(PROJECT_DIR)/lib)

### BOARD_TAG & BOARD_SUB
### It must be set to the board you are currently using. (i.e uno, mega, etc.)
### For the Arduino Uno, only BOARD_TAG is mandatory and BOARD_SUB can be equal to anything
### For the Arduino Mega2560, BOARD_SUB is also needed
BOARD_TAG         = mega
BOARD_SUB         = atmega2560

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 115200

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
AVR_TOOLS_DIR     = /usr

### AVRDDUDE
### Path to avrdude directory.
AVRDDUDE          = /usr/bin/avrdude

### CFLAGS_STD
CFLAGS_STD        = -std=gnu11

### CXXFLAGS_STD
CXXFLAGS_STD      = -std=gnu++11

### CPPFLAGS
### Flags you might want to set for debugging purpose. Comment to stop.
CXXFLAGS         = -pedantic -Wall -Wextra

### If avr-gcc -v is higher than 4.9, activate coloring of the output
ifeq "$(AVR_GCC_VERSION)" "1"
    CXXFLAGS += -fdiagnostics-color
endif

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT      = /dev/tty.usbmodem*

### don't touch this
CURRENT_DIR       = $(shell basename $(CURDIR))

### OBJDIR
### This is were you put the binaries you just compile using 'make'
[... 35 lines cut ...]
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [USER]               CFLAGS_STD = -std=gnu11 
- [USER]               CXXFLAGS_STD = -std=gnu++11 
- [COMPUTED]           DEVICE_PATH = /dev/tty.usbmodem* (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      SYSTEM_LIBS =
- [PLATFORM]             Wire
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.3.4 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------
mkdir -p /home/alex/build/Bare-Arduino-Project/bin/BarProject/mega
/usr/bin/avr-g++ -MMD -c -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=165 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/mega    -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire -I/usr/share/arduino/hardware/arduino/avr/libraries/Wire/utility   -I/home/alex/build/Bare-Arduino-Project/lib/BarLib -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fdiagnostics-color -fno-exceptions -std=gnu++11 BarProject.cpp -o /home/alex/build/Bare-Arduino-Project/bin/BarProject/mega/BarProject.o
In file included from BarProject.cpp:3:0:
/home/alex/build/Bare-Arduino-Project/lib/BarLib/BarLib.h:11:20: fatal error: FooLib.h: No such file or directory
 #include "FooLib.h"
                    ^
compilation terminated.
/home/alex/build/Bare-Arduino-Project/Arduino-Makefile/Arduino.mk:1238: recipe for target '/home/alex/build/Bare-Arduino-Project/bin/BarProject/mega/BarProject.o' failed
make: *** [/home/alex/build/Bare-Arduino-Project/bin/BarProject/mega/BarProject.o] Error 1
```

 I'm using Arch Linux, fully updated, and yep, FooProject compiles just fine.

ladislas, 2015-07-10:
do you have python installed?

ladislas, 2015-07-10:
if you run the following from the `BarProject` directory, what is the output?

```bash
../../Arduino-Makefile/bin/lib-detection ../../lib
```

alexozer, 2015-07-10:
Yep, python and python2 are installed. Arch packages it so that /usr/bin/python is python3, and /usr/bin/python2 is python2. I mention this only because it's caused problems in the past from stuff assuming /usr/bin/python is python2.

Running lib-detection on the lib directory:
```
SKETCH_LIBS
BarLib

SKETCH_LIBS_DEPS
FooLib
```
So it really seems like it's detecting the libraries just fine... so weird.

ladislas, 2015-07-10:
yep, but it should look like this, with stuff on the same line:

```
SKETCH_LIBS BarLib // same line
SKETCH_LIBS_DEPS FooLib // same line
```

Maybe this is causing the bug as the grep is looking for things on one line I think. Try modifying the `lib-detection` script and see if you can get things on the same line.

ladislas, 2015-07-10:
and I even think that you might just need to change the first line with `#!/usr/bin/env python`

alexozer, 2015-07-10:
You're spot on, actually! Changing the first line to python2 makes it compile. The only issue now is how to make this platform-independent (or I could just fork it).

I also notice that the auto-lib branch of arduino-makefile is really far behind the master branch, I wonder if there's any way to either get it merged with master or pull the changes in master into the auto-lib branch.

ladislas, 2015-07-11:
> You're spot on, actually! Changing the first line to python2 makes it compile. The only issue now is how to make this platform-independent (or I could just fork it).

ahah! took me some time but we finally figured it out! :) I think the easiest way is to force the execution of `python2`. `python3` is not available by default on OS X. I'm going to make some tests and see what I can come up with.

> I also notice that the auto-lib branch of arduino-makefile is really far behind the master branch,

yep I have to update that, I'm going to work on it as well ;)

bssstudio, 2015-07-11:
Hello!
Sorry I was not able to test it more thoroughly, I had some stuff going on. 
I can confirm that changing hashbang line in Arudino-Makefile  from `python` to `python2` made it work.

We should open an issue or something to Arduino-Makefile so that it will support distributions with python3 as default 'python'.

ladislas, 2015-07-12:
> We should open an issue or something to Arduino-Makefile so that it will support distributions with python3 as default 'python'.

@bssstudio no not yet. this issue is related only to my fork of Arduino-Makefile and I want to make sure I can fix this before asking them anything ;)

ladislas, 2015-07-14:
@bssstudio @alexozer 
could you try replacing the content of `lib-detection` by the following and tell me what happens?
```python
#!/usr/bin/env python

from __future__ import print_function
import os
import re
import sys

# Set variables
USER_LIB_PATH = sys.argv[1]
USER_LIBS = []

includeRegex = re.compile("(?<=^\#include\s\")(.*)(?=\.h\")", re.DOTALL|re.M)

SKETCH_SRCS = []
SKETCH_LIBS = []

SKETCH_LIBS_DEPS = []
SKETCH_LIBS_DEPS_STACK = []

# Define functions
def outputLibs(libArray):
	for lib in libArray:
		print(lib, end=" ")
	print("")

# Find local sources .ino, .c or .cpp
FILE_END = (".c", ".cpp", ".ino")
SKETCH_SRCS = [f for f in os.listdir(os.curdir) if f.endswith(FILE_END)]

# Find all USER_LIBS
for path, dirs, files in os.walk(USER_LIB_PATH):
	for d in dirs:
		USER_LIBS.append(d)

# Find SKETCH_LIBS included in SKETCH_SRCS
for src in SKETCH_SRCS:
	currentFile = open(src)

	for line in currentFile:
		match = includeRegex.search(line)
		if match is not None:
			group = match.group(1)
			if group in USER_LIBS:
				SKETCH_LIBS.append(group)

SKETCH_LIBS = sorted(set(SKETCH_LIBS))

# Find SKETCH_LIBS_DEPS includes in SKETCH_LIBS
for lib in SKETCH_LIBS:
	if lib in USER_LIBS:
		currentFile = open(os.path.join(USER_LIB_PATH, lib, lib + ".h"))

		for line in currentFile:
			match = includeRegex.search(line)
			if match is not None:
				group = match.group(1)
				if group in USER_LIBS and group not in SKETCH_LIBS:
					SKETCH_LIBS_DEPS_STACK.append(group)

SKETCH_LIBS_DEPS_STACK = list(set(SKETCH_LIBS_DEPS_STACK))

# Recursively find all dependencies of every libraries in USER_LIB_PATH
while len(SKETCH_LIBS_DEPS_STACK) > 0:
	for lib in SKETCH_LIBS_DEPS_STACK:
		if lib in USER_LIBS:
			currentFile = open(os.path.join(USER_LIB_PATH, lib, lib + ".h"))

			for line in currentFile:
				match = includeRegex.search(line)
				if match is not None:
					group = match.group(1)
					if group in USER_LIBS and group not in SKETCH_LIBS_DEPS_STACK and group not in SKETCH_LIBS_DEPS and group not in SKETCH_LIBS:
						SKETCH_LIBS_DEPS_STACK.append(group)

				else:
					if lib not in SKETCH_LIBS_DEPS:
						SKETCH_LIBS_DEPS.append(lib)
						if lib in SKETCH_LIBS_DEPS_STACK:
							SKETCH_LIBS_DEPS_STACK.remove(lib)

SKETCH_LIBS_DEPS.sort()

# Output libraries for the Makefile
print("SKETCH_LIBS", end=" ")
outputLibs(SKETCH_LIBS)

print("SKETCH_LIBS_DEPS", end=" ")
outputLibs(SKETCH_LIBS_DEPS)
```

Or apply the following patch:

```diff
diff --git a/bin/lib-detection b/bin/lib-detection
index 8753ecd..be9caee 100755
--- a/bin/lib-detection
+++ b/bin/lib-detection
@@ -1,10 +1,10 @@
 #!/usr/bin/env python

+from __future__ import print_function
 import os
 import re
 import sys

-
 # Set variables
 USER_LIB_PATH = sys.argv[1]
 USER_LIBS = []
@@ -21,7 +21,7 @@ SKETCH_LIBS_DEPS_STACK = []
 # Define functions
 def outputLibs(libArray):
 	for lib in libArray:
-		print(lib),
+		print(lib, end=" ")
 	print("")

@@ -83,8 +83,8 @@ while len(SKETCH_LIBS_DEPS_STACK) > 0:
 SKETCH_LIBS_DEPS.sort()

 # Output libraries for the Makefile
-print("SKETCH_LIBS"),
+print("SKETCH_LIBS", end=" ")
 outputLibs(SKETCH_LIBS)

-print("SKETCH_LIBS_DEPS"),
-outputLibs(SKETCH_LIBS_DEPS)
\ No newline at end of file
+print("SKETCH_LIBS_DEPS", end=" ")
+outputLibs(SKETCH_LIBS_DEPS)
```

ladislas, 2015-07-14:
@alexozer I just updated the repo to pull all the recent improvements from sudar's arduino makefile.

let me know if you have any issues :)

alexozer, 2015-07-14:
@ladislas Thanks for your work on this, the latest Bare-Arduino-Project works with the patch you posted. So I assume you're going to commit the patch to your forked version of Arduino-Makefile, or try to get it into the original Arduino-Makefile?

ladislas, 2015-07-26:
@alexozer yep! A little more work and it should be okay ;)

closed 2015-10-23 by ladislas

T3KN0KRAT, 2017-03-26:
Yikes! I reeeely hate to resurrect this, but I've followed this thread, tried its suggestions, and am still coming up with the same error: fatal error: FooLib.h: No such file or directory.  

This one, however is on a raspbian install.

Shall I open a new issue?  Or shall I add details here (in another comment)?