ladislas/Bare-Arduino-Project #16: Why does "PROJECT_DIR" need to be set?

dgtlmoon, 2015-05-21:
Great project! why do I need to set this variable in my Makefile? This would make it completely un-portable

I've tried setting it to just 

```PROJECT_DIR       = .```

However i get the following output when following the INSTALL.md

```dgtlmoon@dgtlmoon-ThinkPad-T420s:/tmp/new/src/BarProject $  [master|✔]make
Makefile:73: Arduino-Makefile/Arduino.mk: No such file or directory
make: *** No rule to make target `Arduino-Makefile/Arduino.mk'.  Stop.```

dgtlmoon, 2015-05-21:
The error is caused by ARDMK_DIR being incorrect in my case once PROJECT_DIR is set.. Is there a more dynamic way todo it than below? something more portable?

```
PROJECT_DIR       = .
ARDMK_DIR         = ../../Arduino-Makefile
```

ladislas, 2015-05-21:
> why do I need to set this variable in my Makefile? This would make it completely un-portable

because people don't always set up their computer the same way and might have things in different places :)

setting `PROJECT_DIR = .` will set the `root` where the `Makefile` is called.

It certainly is possible to do this in a more dynamic way, if you want to give it a try, I'll be happy to merge it :)

closed 2015-05-21 by dgtlmoon