ladislas/Bare-Arduino-Project #17: Getting a synch error when trying to upload a sketch

saqib-zmi, 2015-05-30:
I'm a Mac 10.10 User trying to free myself from the terrible Arduino IDE. I'm using an Arduino UNO board.

I followed the instructions here: https://github.com/ladislas/Bare-Arduino-Project/blob/master/INSTALL.md#4-test-that-evrything-is-up-and-running

However, when I did "make upload", I got the following error:

    mkdir -p /Users/syed.saqibali/Arduino/Oryx/bin/BarProject/mega
    /Applications/Xcode.app/Contents/Developer/usr/bin/make reset
    /Users/syed.saqibali/Arduino/Oryx/Arduino-Makefile/bin/ard-reset-arduino  /dev/tty.usbmodem1451
    /Applications/Xcode.app/Contents/Developer/usr/bin/make do_upload
    /usr/local/bin/avrdude -q -V -p atmega1280 -D -c arduino -b 57600 -P /dev/tty.usbmodem1451 \
			-U flash:w:/Users/syed.saqibali/Arduino/Oryx/bin/BarProject/mega/BarProject.hex:i
    avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xbc
    avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xef
    avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xf8
    avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xae
    avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3d
    avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xf8
    avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xbc
    avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xad
    avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xf8
    avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xae

I don't think this is a problem with the board itself because when I open the IDE and try to upload a compiled sketch through it, it works fine

ladislas, 2015-05-30:
It looks like `avrdude` thinks you are using an Arduino Mega1280: 

```
/usr/local/bin/avrdude -q -V -p **atmega1280** -D -c arduino -b 57600 -P /dev/tty.usbmodem1451 \
        -U flash:w:/Users/syed.saqibali/Arduino/Oryx/bin/BarProject/mega/BarProject.hex:i
```

Could you copy/paste me your `Makefile` please? :)

saqib-zmi, 2015-05-31:
Yes. Thank you! My makefile was faulty.

It had this in there:

    ### BOARD_TAG & BOARD_SUB
    ### It must be set to the board you are currently using. (i.e uno, mega, etc.)
    ### For the Arduino Uno, only BOARD_TAG is mandatory and BOARD_SUB can be equal to anything
    ### For the Arduino Mega2560, BOARD_SUB is also needed
    BOARD_TAG         = mega
    BOARD_SUB         = atmega2560

When I changed "mega" to "uno" the upload succeeded.

closed 2015-05-31 by saqib-zmi

ladislas, 2015-05-31:
glad you found the error :+1: