ladislas/Bare-Arduino-Project #18: unrecognized CFLAGS and CXXFLAGS

romanchyla, 2015-07-04:
Hi,
Thank you for putting the sample project together, I'd like to report a problem on Ubuntu 14.04 LTS

```
CFLAGS_STD        = -std=gnu11
CXXFLAGS_STD      = -std=gnu++11
```

are not recognized options, compilation failed.

```
/usr/bin/avr-gcc -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=100  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/standard    -Wall -ffunction-sections -fdata-sections -Os -std=gnu11 /usr/share/arduino/hardware/arduino//cores/arduino/WInterrupts.c -o /dvt/workspace/company/hw/arduino/bin/BarProject/uno/core/WInterrupts.o
cc1: error: unrecognized command line option "-std=gnu11"
make: *** [/dvt/workspace/company/hw/arduino/bin/BarProject/uno/core/WInterrupts.o] Error 1

```

I have updated my toolchain to gcc4.8 (and also g++, cpp) 
http://askubuntu.com/questions/271388/how-to-install-gcc-4-8

But the problem persists, do you think it is because of an old avr-gcc? Is there a way around it? (the build from source seems rather daunting)

```
avr-gcc -v
Using built-in specs.
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/avr/4.5.3/lto-wrapper
Target: avr
Configured with: ../src/configure -v --enable-languages=c,c++ --prefix=/usr/lib --infodir=/usr/share/info --mandir=/usr/share/man --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --enable-shared --with-system-zlib --enable-long-long --enable-nls --without-included-gettext --disable-libssp --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=avr
Thread model: single
gcc version 4.5.3 (GCC) 
```

I have solved 'my problem' by editing the flags in the Makefile. 

```
CFLAGS_STD        =
CXXFLAGS_STD      = -std=gnu++0x
```

It then compiles and uploads/works well. BUT I'm not sure what these flags do and if it is safe to ignore them - according to https://gcc.gnu.org/gcc-4.7/changes.html, the gnu++0x is equivalent of gnu++11 (so it should be safe), but I don't understand C and the gnu11. 

Can you please shed some light on it?

romanchyla, 2015-07-04:
this helped me fill the gap: https://github.com/WeAreLeka/Arduino-Makefile/blob/master/arduino-mk-vars.md#cxxflags_std

I think it might be useful adding warning directly into the Makefile, next to the CXXFLAGS.... something like:

Values depend on the installed versions of your gcc/c++/avr-gcc and the code that is being compiled
For details, see: https://github.com/WeAreLeka/Arduino-Makefile/blob/master/arduino-mk-vars.md#cxxflags_std

ladislas, 2015-07-04:
Hi @romanchyla,

Thanks for reporting this issue :+1: You're right, I will add a warning about the flags regarding the `-std=` option.

On Ubuntu, `gcc` and `avr-gcc` are independent. And your version `avr-gcc 4.5.3` is rather old. To use `avr-gcc 4.8.x` you must run the following:

```
$ sudo apt-get update && sudo apt-get upgrade
$ sudo apt-get install gcc-avr binutils avr-libc avrdude
```

And you should have the following:

[image]

romanchyla, 2015-07-05:
oups, you are right! i've started on ubuntu 11 and did the upgrade, but I haven't verified that i was on 14.04 (and I wasn't! silly old me...). i can confirm that it now works with the gnu11 options, well at least I learnt somehting extra about C standards :)

closed 2015-07-05 by ladislas