ladislas/Bare-Arduino-Project #19: YouCompleteMe not properly using Teensy/Teensyduino headers

alexozer, 2015-07-10:
I'm trying to set up YouCompleteMe (ycm) in Vim to autocomplete and error-detect code for Teensy 3.1. I installed Arduino 1.0.6 to a separate directory (because Bare-Arduino-Project doesn't support Arduino 1.6.* for Teensy), and installed Teensyduino to it. Then, I pointed ycm to this Arduino directory via .ycm_extra_conf.py:
```
ArduinoLibDir = [
  "/home/alex/build/arduino-1.0.6" // location of Arduino patched with Teensyduino
  ,"/Applications/Arduino.app/Contents/Resources/Java/libraries"
  ,"/Applications/Arduino.app/Contents/Java/libraries"
  ,"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries"
]
```
When using this test code:
```
#include <Arduino.h>

constexpr int pin = 13;

int main() {
	pinMode(pin, OUTPUT);
	digitalWrite(pin, HIGH);
	delay(1000);

	Serial.begin(9600);
	Serial.println("There, I turned on the LED, see?");
}
```
ycm throws no errors with pinMode(), digitalWrite(), and delay(), indicating it found them in Arduino.h just fine, but it doesn't recognize the Serial object:
[image]

Yet strangely, if I pretend I'm writing code for an original Arduino by changing the ycm library path to an unpatched Arduino directory:
```
ArduinoLibDir = [
  "/usr/share/arduino" // unpatched arduino
  ,"/Applications/Arduino.app/Contents/Resources/Java/libraries"
  ,"/Applications/Arduino.app/Contents/Java/libraries"
  ,"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries"
]
```
ycm recognizes Serial and autocompletes it just fine.

So, this leads me to believe that there's something to be fixed with the supplied .ycm_extra_conf.py.

ladislas, 2015-07-11:
>  (because Bare-Arduino-Project doesn't support Arduino 1.6.* for Teensy)

no it's because Arduino-Makefile doesn't support the Teensy. There is a lot of changes going on in the way the Arduino IDE works and on how to parse the `board.txt` and so on. Until everyone is settled down, you have to use the `v1.0.6` for the teensy to work correctly with the `Makefile`

I'm going to try and find a solution.

For the moment, as the Teensy uses all the same functions as Arduino, you can debug your code with the unpatched version :)

alexozer, 2015-07-12:
Agh, alright, I'll just deal with it.

closed 2015-12-16 by ladislas

tiagoboldt, 2017-08-19:
Hey guys, afaik, this is still an issue today. @ladislas any reason to close it?

alexozer, 2017-08-19:
TBH there's better ways to get completion nowadays, like with platformio for vscode.

ladislas, 2017-08-19:
@tiagoboldt I don't use the teensy anymore. I don't have much time to investigate but I'd be happy to review any pull request.

I guess it's just about tweaking the ycm config :)