ladislas/Bare-Arduino-Project #21: Problems with dependencies with different folder names

Limpinho0, 2015-09-24:
Hi, I'm newbie on python, and I have a complex arduino program that uses some libraries in just one folder. I guess your program works taking the names of the folders and using to know the name of the libraries on it. But I have many libraries on just one folder of the lib, and the software can't include this files on the project. I tried put one folder for each files but some files includes other libraries. Can you help me with it? Thanks

ladislas, 2015-09-26:
yep, that's something I've been wanting to work on for a few months but I haven't got the time yet.

I'll try to get it fixed ASAP ;)

jon-whit, 2015-09-30:
I would also like to do this. Here is what I am trying to do:

lib/Station.h
-- #include <XBee.h>
lib/PowerManager.h

src/H2OiQ.cpp
-- #include "Station.h"
-- #include "PowerManager.h"

The `libdetection.py` script that you added to the Arduino-Makefile project doesn't allow this. One additional thing to note in addition to your suggested enhancement above:

* Arduino system libraries (Arduino.h, SoftwareSerial.h, XBee.h, etc..) cannot be included in library files. 

 Is this an easy fix? I'd love to utilize this project for my next project.

ladislas, 2015-09-30:
> Arduino system libraries (Arduino.h, SoftwareSerial.h, XBee.h, etc..) cannot be included in library files.

@jon-whit: what do you mean? what system are you on?

jon-whit, 2015-09-30:
In BarProject.cpp if you include SoftwareSerial.h you don't get any compilation errors:

```
#include <SoftwareSerial.h>

void setup() {
...
}

void loop() {
...
}
```

However, if you try to include the same SoftwareSerial system header in BarLib.h and then include BarLib.h in BarProject.cpp, then compilation fails with message:

```
In file included from BarProject.cpp:3:0:
/home/jonathan/GitHub/Bare-Arduino-Project/lib/BarLib/BarLib.h:11:28: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
```

ladislas, 2015-10-23:
@jon-whit my apologies for the long delay, I've been pretty busy recently with my startup.

I tried what you said, including `#include <SoftwareSerial.h>` in the `BarProject` and `BarLib` and everything compiles just fine.

Also, if you look at the content of the library files, you'll see that I have included `Arduino.h`.

Could you try cloning the project again and tell me if it still doesn't work?

mfe5003, 2015-11-06:
I am having a similar problem as @jon-whit, and I just cloned the project a day or two ago.
I haven't had a chance to look into it yet but I suspect that the code that generates the included libraries is only looking at the top level file i.e. `BarProject.cpp`

The work around I'm using, is that if I want to add `SPI.h` in a library I made I need to also add it in the top level file, even if I don't need it there.

So if in `BarLib.h` I want:
`#include <SPI.h>`

I need to also put in `BarProject.cpp`:
`#include <SPI.h>`

as well.

ladislas, 2015-11-07:
yep that's right -- for Arduino libs, you need to include them in your project.

jon-whit, 2015-12-13:
@ladislas This issue still has yet to get fixed. The `libdectection.py` script that you created is not inclusive enough for #includes directives. Consider the simple Arduino sketch directory shown below:

```
/home/user/sketchbook
.... readme.txt
.... xbee_lib
........ build.xml
........ docs
........ examples
........ XBee.h
........ XBee.c
```

If you have a project file `main.cpp` in `src` that includes `XBee.h`,

```
// main.cpp
#include <Arduino.h>
#include <XBee.h>

void setup()
{
    Serial.begin(9600);
    XBee xbee = XBee();
    xbee.setSerial(Serial);
}

void loop()
{
}
```

Running the compilation target in the Makefile fails with an error indicating that XBee.h could not be found.
```
> avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/standard -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fno-exceptions -std=gnu++11 main.cpp -o /home/user/dir/project-dir/bin/project-dir/uno/main.cpp.o
main.cpp:2:X: fatal error: XBee.h: No such file or directory
 #include <XBee.h>
```

I'm simply trying to use 3rd party libraries for some of my Arduino projects, but the only way I have found to do so is to copy the 3rd party library into the `lib` directory and change the headers and definition file names to names of the form xxxxLib.(h,c).

ladislas, 2015-12-13:
@jon-whit : do you have a repository on github that I could clone to test your settings.

I think I already raised an issue like the one you're having here: https://github.com/sudar/Arduino-Makefile/issues/249#issuecomment-54158707

There are two things that might get in the way of doing what you want:

- the first is the way the Makefile works: it is based on how Arduino works and doesn't allow you to put all your libraries in a big `lib` folder and simply include them. It wants you to do as Arduino says, i.e. have a folder named `LibName` with inside you files `LibName.h` and `LibName.cpp`. You can also have other files with different name but you'll only be able to reference the one with `LibName` in your `main.cpp`
- the second is that the Makefile does a lot of checking and processing to be as easy to use as possible for beginners. But it uses the `Makefile` language which is complicated and prone to errors. There is an idea of rewriting the full logic in python to be more easily customizable, still using `make` to call everything, but it will take time

my script allow you to not reference all the the headers you need to compile your sketch. It looks for other headers in your `lib/LibName` folders and include them automatically. It does not however look for header outside of the clone repo. Everything must be in `lib`. My project Moti is a good exemple of how things should work (https://github.com/WeAreLeka/moti). You can just clone it, customize the Makefile with your own paths and it should work.

I do not plan on adding the possibility to look for libs outside of your project as the initial purpose of the script was to be able to package everything in the same place and be able to share it with a team, without having to install outside libraries in complex locations.

ladislas, 2015-12-16:
any news on this?

ladislas, 2016-01-24:
i'm closing this issue, feel free to reopen if needed.

closed 2016-01-24 by ladislas