ladislas/Bare-Arduino-Project #22: BOARD_TAG & BOARD_SUB

jej, 2015-12-02:
Hi,
I had some troubles to set up BOARD_TAG & BOARD_SUB in the Makefile. May I suggest to improve INSTALL.md?
Here is what I did to set up the good values:
* Open the BOARDS_TXT file (for example /Users/me/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt) as specified when 'make' is launched.
* Find for the board used (for example *"Arduino Pro or Pro Mini (3.3V, 8 MHz) w/ ATmega328"*)
* Look at the config keys for this board (in this case "*pro*.menu.cpu.*8MHzatmega328*=ATmega...")
* So BOARD_TAG=*pro* and BOARD_SUB=*8MHzatmega328*.

*make* & enjoy :) Hope it helps...

ladislas, 2015-12-02:
Hi @jej -- thank you for your contribution.

can you make a pull request? also, it might be good in the instruction to say that `BOARD_SUB` is only used in the most recent versions of the IDE. and not in `1.0.x`

thanks!

jej, 2015-12-02:
Ok! I will do it soon.
J.

ladislas, 2015-12-02:
fixed by #23

closed 2015-12-02 by ladislas

jej, 2015-12-02:
Need to update the TOC, but I guess I cannot do it myself...
J.

ladislas, 2015-12-02:
just did it :)

jej, 2015-12-02:
well done then! thanks.