ladislas/Bare-Arduino-Project #24: Doesn't build objects from sources under lib/

sagargp, 2015-12-04:
I have some custom libraries under the lib/ directory, similar to FooLib and BarLib. When building, the compiler is invoked with -I.../FooLib properly, but a FooLib.o is never built and linked into the final executable. Therefore, the final linker step fails with "undefined references" errors.

ladislas, 2015-12-04:
can you show me your directory structure?
also, does your library have the same name as your folder?

if you have a github repository with everything, that would be even better for me to take a look.

sagargp, 2015-12-04:
They do have the same name, although the directory/file names are in all-lowercase while the class name is in CamelCase. I don't know if that matters. [Here's a link to the code I'm talking about.](https://github.com/sagargp/hermes2-firmware/tree/98b89c0c86cf2048d0648aae843e73d3eb057b61)

ladislas, 2015-12-05:
this exemple works

https://github.com/ladislas/hermes2-firmware/tree/hermes

ladislas, 2015-12-05:
you need to change the makefile a bit

ladislas, 2015-12-16:
@sagargp any news on that?

I'm closing the issue, feel free to reopen it if needed.

closed 2015-12-16 by ladislas