ladislas/Bare-Arduino-Project #25: Single library across multiple projects

g770, 2015-12-15:
Is it possible to have a single library that is shared across projects?  In playing with this it looks like there is a 1:1 mapping between project and library, for example FooProject and FooLib.

ladislas, 2015-12-15:
if your library is in `lib/myLib`, you can reference it from every project in `src/myProject` and have multiple project reference the same library.

but it can not be outside of the whole project. if you want to do so you can use submodules.

g770, 2015-12-15:
Hmm... ok, let me plan around a bit.

ladislas, 2015-12-16:
my repo https://github.com/WeAreLeka/moti is a good example on how this works.

you'll see that in `src` I have different robot's behavior that all use the same libraries included from `lib`

I'm closing this now, feel free to reopen if needed.

closed 2015-12-16 by ladislas