ladislas/Bare-Arduino-Project #26: External libraries do not pick up

ilvadol, 2016-01-21:
I'm trying to build a project that uses F Malpartida's  New LiquidCrystal libraries (```LCD.h and LiquidCrystal_I2C.h```). I put those into lib of the project folder as suggested. I'm also using FooProject folder, as it already contains a Makefile after I run the check that everything is working.
When I "make" my project the terminal goes on computing and so on and then:
```
mkdir -p /Users/ilvadol/ardprj/bin/FooProject/uno
/usr/local/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=167 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard    -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/utility  -Wall -ffunction-sections -fdata-sections -Os -pedantic -Wall -Wextra -fdiagnostics-color -fno-exceptions -std=gnu++11 FooProject.cpp -o /Users/ilvadol/ardprj/bin/FooProject/uno/FooProject.cpp.o
FooProject.cpp:3:17: fatal error: LCD.h: No such file or directory
 #include "LCD.h"
                 ^
compilation terminated.
make: *** [/Users/ilvadol/ardprj/bin/FooProject/uno/FooProject.cpp.o] Error 1
```
I tried to  ```#include <LCD.h>``` but it doesn't work.
My Makefile is unchanged since the initial check. What am I doing wrong?
I'm on a MacOSX 10.11.2
Using MacVIM 7.4.1147
Arduino IDE v.1.6.7
Trying to run on UNO.

Thanks a lot in advance.

ilvadol, 2016-01-21:
Or should I create a new folder with a new reconfigured Makefile?

ladislas, 2016-01-22:
hi @ilvadol !

in `./lib` you have to add your libraries in folders with the same name as the `.h` and `.cpp` files if you want them to be referenced.

for example here, I include the `Stabilization.h` librairie:

- main sketch: https://github.com/weareleka/moti/blob/dev/src/Stabilization/main.cpp
- lib: https://github.com/weareleka/moti/blob/dev/lib/Stabilization/Stabilization.h

ilvadol, 2016-01-22:
Thanks for the answer, @ladislas
That's great. I'll check it later for sure. But could you please make it clear somewhere in ```.lib/Readme.mk``` Because at the moment, it's not very clear how you put bundles of libraries to get indexed.

ilvadol, 2016-01-22:
Hmm... Now it's more interesting. Putting the files into folders with the same name partly solves the problem. BUT when there are NO ```.h``` in the ```/FooProject``` I get this error:
```
mkdir -p /Users/ilvadol/ardprj/bin/FooProject/uno
make: *** No rule to make target `LCD.h', needed by `/Users/ilvadol/ardprj/bin/FooProject/uno/FooProject.cpp.o'.  Stop.
```
But when there are ```.h``` files in the ```/FooProject`` folder, it works, compiles and uploads.

ladislas, 2016-01-22:
can you push your project on github so that I can try it myself?

ilvadol, 2016-01-22:
well... I can... but because I'm newbie here (so sorry for that), I do not know how. I read the guides and learn as i go. So anyway, I try to push using guides and it does not include any of my work. I can attach ```ardprj.zip``` file of my entire project folder. 
[ardprj.zip](https://github.com/ladislas/Bare-Arduino-Project/files/101121/ardprj.zip)
And ```Archive.zip``` contains the ```.h``` files that need to go into ```FooProject``` in order for it to upload. 
[Archive.zip](https://github.com/ladislas/Bare-Arduino-Project/files/101119/Archive.zip)

ilvadol, 2016-01-23:
Sorry for the wait and the possible confusion that it might have caused you (I'm not very familiar with CLI as I don't have much experience). But there are GUI that github provides. 
Here it is, my repo:
https://github.com/ilvadol/ardprj

ladislas, 2016-01-24:
can you copy paste your Makefile here? also, what version of the Arduino IDE are you using?

ilvadol, 2016-01-24:
Sure. Here it is:
```
### DISCLAIMER
### This is an example Makefile and it MUST be configured to suit your needs.
### For detailled explanations about all the avalaible options,
### please refer to https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md

### PROJECT_DIR
### This is the path to where you have created/cloned your project
PROJECT_DIR       = /Users/ilvadol/ardprj
# PROJECT_DIR       = $(HOME)/dev/arduino/bare-arduino-project

### AVR_GCC_VERSION
### Check if the version is equal or higher than 4.9
AVR_GCC_VERSION  := $(shell expr `avr-gcc -dumpversion | cut -f1` \>= 4.9)

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(PROJECT_DIR)/Arduino-Makefile

### ARDUINO_DIR
### Path to the Arduino application and ressources directory.
### For Arduino IDE 1.0.x
# ARDUINO_DIR       = /Applications/Arduino.app/Contents/Resources/Java
### For Arduino IDE 1.6.x
ARDUINO_DIR       = /Applications/Arduino.app/Contents/Java

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
USER_LIB_PATH     :=  $(PROJECT_DIR)/lib

### BOARD_TAG & BOARD_SUB
### For Arduino IDE 1.0.x
### Only BOARD_TAG is needed. It must be set to the board you are currently using. (i.e uno, mega2560, etc.)
# BOARD_TAG         = mega2560
### For Arduino IDE 1.6.x
### Both BOARD_TAG and BOARD_SUB are needed. They must be set to the board you are currently using. (i.e BOARD_TAG = uno, mega, etc. & BOARD_SUB = atmega2560, etc.)
### Note: for the Arduino Uno, only BOARD_TAG is mandatory and BOARD_SUB can be equal to anything
BOARD_TAG         = uno

BOARD_SUB         = atmega2560

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 115200

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
AVR_TOOLS_DIR     = /usr/local

### AVRDDUDE
### Path to avrdude directory.
AVRDDUDE          = /usr/local/bin/avrdude

### CFLAGS_STD
CFLAGS_STD        = -std=gnu11

### CXXFLAGS_STD
CXXFLAGS_STD      = -std=gnu++11

### CPPFLAGS
### Flags you might want to set for debugging purpose. Comment to stop.
CXXFLAGS         = -pedantic -Wall -Wextra

### If avr-gcc -v is higher than 4.9, activate coloring of the output
ifeq "$(AVR_GCC_VERSION)" "1"
    CXXFLAGS += -fdiagnostics-color
endif

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT      = /dev/tty.usbmodem*

### OBJDIR
### This is were you put the binaries you just compile using 'make'
CURRENT_DIR       = $(shell basename $(CURDIR))
OBJDIR            = $(PROJECT_DIR)/bin/$(CURRENT_DIR)/$(BOARD_TAG)

### path to Arduino.mk, inside the ARDMK_DIR, don't touch.
include $(ARDMK_DIR)/Arduino.mk
```
It's nothing unusual, really.

ladislas, 2016-01-24:
thanks.

I does work on my computer. I just had to clone the repo, add the makefile and everything compiles fine.

ilvadol, 2016-01-24:
So, even without ```LCD.h```?

ladislas, 2016-01-24:
with LCD.h inside `./lib/LCD`

ilvadol, 2016-01-24:
well... For me it does not... 
```
mkdir -p /Users/ilvadol/ardprj/bin/testProject/uno
make: *** No rule to make target `LCD.h', needed by `/Users/ilvadol/ardprj/bin/testProject/uno/FooProject.cpp.o'.  Stop.
```
Do not really know why? It's fresh installed OSX. Strange. Just a sec. Have to change to linux (raspberry pi's jessie) and check it there.

ilvadol, 2016-01-24:
On the PI it compiles and uploads even without ```LCD.h```
But on the OSX it asks for it.
Of course, i did change ```makefiles``` to suit the system. But it's strange, anyhow.

ladislas, 2016-01-24:
I did some cleanup here: https://github.com/ladislas/ardprj/tree/remove-libs

I removed all the libs and only added back the ones needed to compile in `./lib/LCD`. You can put everything in the same folder as the compiler will look for them automatically in the same folder.

It compiles on OSX

ilvadol, 2016-01-24:
Well... I guess it's incurable. When i deleted all libs and put everything as you suggested I got this:
```
/Users/ilvadol/ardprj/Arduino-Makefile/Arduino.mk:960: *** The following libraries specified in ARDUINO_LIBS could not be found (searched USER_LIB_PATH, ARDUINO_LIB_PATH and ARDUINO_PLATFORM_LIB_PATH): LiquidCrystal_I2C.  Stop.
```
When I put them back in folders it asks for a copy of ```LCD.h``` to be in the ```testProject```:
```
make: *** No rule to make target `LCD.h', needed by `/Users/ilvadol/ardprj/bin/testProject/uno/FooProject.cpp.o'.  Stop.
```
On Linux it works without the problems...
I guess, there is something to do with my OSX setup and how it handles this particular library. Anyway, thanks a lot for your help.

ladislas, 2016-01-24:
we're trying one last thing, just follow the following (don't copy the `$`):

```shell
# open terminal
$ cd $HOME
$ mkdir dev
$ cd dev

# clone repository
$ git clone https://github.com/ladislas/ardprj
$ cd ardprj
$ git submodule update --init --recursive

# change branch
$ git checkout remove-libs

# change dir to src/project
$ cd src/LCDProject

# copy and modify the makefile
$ cp ../../Makefile-OSX.mk ./Makefile

# compile
$ make
```

it should work.

ilvadol, 2016-01-24:
Unfortunately, i did exactly what you told me to do, and it did not work...
```
/Users/ilvadol/dev/ardprj/Arduino-Makefile/Arduino.mk:960: *** The following libraries specified in ARDUINO_LIBS could not be found (searched USER_LIB_PATH, ARDUINO_LIB_PATH and ARDUINO_PLATFORM_LIB_PATH): LiquidCrystal_I2C.  Stop.
```
That's what it said.
Here is my makefile:
```
### DISCLAIMER
### This is an example Makefile and it MUST be configured to suit your needs.
### For detailled explanations about all the avalaible options,
### please refer to https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md

### PROJECT_DIR
### This is the path to where you have created/cloned your project
PROJECT_DIR       = /Users/ilvadol/dev/ardprj
# PROJECT_DIR       = $(HOME)/dev/arduino/bare-arduino-project

### AVR_GCC_VERSION
### Check if the version is equal or higher than 4.9
AVR_GCC_VERSION  := $(shell expr `avr-gcc -dumpversion | cut -f1` \>= 4.9)

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(PROJECT_DIR)/Arduino-Makefile

### ARDUINO_DIR
### Path to the Arduino application and ressources directory.
### For Arduino IDE 1.0.x
# ARDUINO_DIR       = /Applications/Arduino.app/Contents/Resources/Java
### For Arduino IDE 1.6.x
ARDUINO_DIR       = /Applications/Arduino.app/Contents/Java

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
USER_LIB_PATH     :=  $(realpath $(PROJECT_DIR)/lib)

### BOARD_TAG & BOARD_SUB
### For Arduino IDE 1.0.x
### Only BOARD_TAG is needed. It must be set to the board you are currently using. (i.e uno, mega2560, etc.)
# BOARD_TAG         = mega2560
### For Arduino IDE 1.6.x
### Both BOARD_TAG and BOARD_SUB are needed. They must be set to the board you are currently using. (i.e BOARD_TAG = uno, mega, etc. & BOARD_SUB = atmega2560, etc.)
### Note: for the Arduino Uno, only BOARD_TAG is mandatory and BOARD_SUB can be equal to anything
BOARD_TAG         = uno
BOARD_SUB         = atmega2560

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 115200

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
AVR_TOOLS_DIR     = /usr/local

### AVRDDUDE
### Path to avrdude directory.
AVRDDUDE          = /usr/local/bin/avrdude

### CFLAGS_STD
CFLAGS_STD        = -std=gnu11

### CXXFLAGS_STD
CXXFLAGS_STD      = -std=gnu++11

### CPPFLAGS
### Flags you might want to set for debugging purpose. Comment to stop.
CXXFLAGS         = -pedantic -Wall -Wextra

### If avr-gcc -v is higher than 4.9, activate coloring of the output
ifeq "$(AVR_GCC_VERSION)" "1"
    CXXFLAGS += -fdiagnostics-color
endif

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT      = /dev/tty.usbmodem*

### OBJDIR
### This is were you put the binaries you just compile using 'make'
CURRENT_DIR       = $(shell basename $(CURDIR))
OBJDIR            = $(PROJECT_DIR)/bin/$(CURRENT_DIR)/$(BOARD_TAG)

### path to Arduino.mk, inside the ARDMK_DIR, don't touch.
include $(ARDMK_DIR)/Arduino.mk
```

Just a second, before it's considered a MY-SYSTEM-SPECIFIC issue i have to check something.

ilvadol, 2016-01-24:
WOW.
I guess it was a MAJOR 3 DAYS LONG GLITCH.
After your advice to make everything from scratch it failed and i decided to put ```.h``` and ```.cpp``` libs inside folders (your initial suggestion) once again
After ```make``` it worked!!!
i can't thank you enough! Thanks a whole lot!

You may close the issue.
And also, please, update the description how libraries are held.
How can I merge everything back to my project? Should i delete my repo and make a new one?

ladislas, 2016-01-24:
you're welcome.

yeah, better make a new repo ;)

closed 2016-01-24 by ladislas