ladislas/Bare-Arduino-Project #28: Arduino 1.6.9 only (without having to brew avrdude and avr-libc)

kenji21, 2016-06-21:
Not really an issue, but an "interesting feature" : only depend on Arduino.app.

I've sucessfully compiled and uploaded the blink example on OS X without brewing avr-libc and avrdude.

To use the one embedded in Arduino.app :
````
AVR_TOOLS_DIR     = /Applications/Arduino.app/Contents/Java/hardware/tools/avr
AVRDDUDE          = /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude
````

In Arduino-Makefile :
````
$ ln -s /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf avrdude.conf
$ ls -l |grep dude
avrdude.conf -> /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf

$ cd examples/Blink
$ make
````

ladislas, 2016-09-25:
@kenji21 thanks for opening this issue.

yes it is absolutely possible to compile by relying only on arduino (it's working with 1.6.12).

but the idea behind the project was to be able to use a more recent version of avr-gcc and get all the cutting edge feature :)

closed 2016-09-25 by ladislas

kenji21, 2016-09-25:
Ok, thanks for taking time to reply