ladislas/Bare-Arduino-Project #30: GCC 6.x causes delays to hang forever

nathansizemore, 2016-09-27:
[Original, mis-filed issue](https://github.com/sudar/Arduino-Makefile/issues/458).

I'm assuming this is also the case for any GCC above 4.9, but have only confirmed with 6.1 and 6.2.

See original issue for details of makefile and such.

Toolchain recreation info:

**Installed Versions**

```
binutils....v2.26.0
gcc.........v6.1.0
avr-libc....v2.0.0

gmp.........v6.1.1
isl.........v0.16.1
mpc.........v1.0.3
mpfr........v3.1.4
cloog.......v0.18.1
```

**binutils configuration**

```
--target=avr \
    --disable-nls \
    --disable-werror
```

**gcc configuration**

(After gmp and friends placed in tree)

```
--target=avr \
    --enable-languages=c,c++ \ 
    --disable-nls \
    --with-gnu-ld \
    --with-dwarf2
```

**libc configuration**

```
--host=avr
```

ladislas, 2016-09-28:
can you provide me with a github repo i could clone and test it myself. 

haven't had any issues with my own projects.

ladislas, 2016-09-28:
I confirm i can compile and upload a code with `delay()` and it works (if that was the problem).

i think your issue, from what i've seen is that you `return 0;` at the end of your `main()` so your program just stops.

ladislas, 2016-09-28:
Okay, I found the error.

First off, you need to `init()` when just using `main()`. Second, your delay was too short between the past print and the return.

So here you go:

```cpp
#include <Arduino.h>

int main(void) {

	init();

	Serial.begin(115200);
	delay(100);

	Serial.println("asdf1");
	delay(10);
	Serial.println("asdf2");
	delay(10);

	return 0;
}
```

closed 2016-09-28 by ladislas