ladislas/Bare-Arduino-Project #33: No module named serial

String-bikinis, 2017-01-25:
First off, I know there was an earlier issue like this one but I have tried everything that was suggested.  When I try to upload using "make upload" I get an error saying "there is no module named serial".

    /Users/j*********/MyArduinoProject/Arduino-Makefile/bin/ard-reset-arduino       /dev/tty.usbmodem1451
    Traceback (most recent call last):
      File "/Users/j*********/MyArduinoProject/Arduino-Makefile/bin/ard-reset-arduino", line 4, in <module>
        import serial
    ImportError: No module named serial
    make[1]: *** [reset] Error 1
    make: *** [upload] Error 2

But I am able to import serial in the python 2.7 interpreter.

    Python 2.7.9 (default, Dec 30 2014, 18:28:09) 
    [GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.56)] on darwin
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import serial
    >>> 

I'm completely lost on this because the 'make upload' scripts should be using python 2.7.  Please help me on this.

ladislas, 2017-01-25:
what happens if you run `which python` in your terminal?

also, can you try running the `ard-reset-arduino` from the terminal?

String-bikinis, 2017-01-25:
$ which python
    /Library/Frameworks/Python.framework/Versions/2.6/bin/python
So the default python version for my computer is 2.6. Is this the version that all bash scripts like ard-reset-arduino use then? I get the same error when I just run the ard-reset-arduino:
    $ open ard-reset-arduino
    $ /Users/j********/MyArduinoProject/Arduino-Makefile/bin/ard-reset-arduino ; exit;
    Traceback (most recent call last):
      File "/Users/j********/MyArduinoProject/Arduino-Makefile/bin/ard-reset-arduino", line 4, in <module>
        import serial
    ImportError: No module named serial

String-bikinis, 2017-01-25:
So i'm not sure how change the default python version to 2.7. Even if I did I wouldn't want to do that since there must be a few applications that rely on the default 2.6 python libraries.  I tried to add a few new directories to the search path in the ard-reset-arduino script before it imports serial.

    import sys
    sys.path.insert(0, '/usr/local/lib/python2.7')
    sys.path.insert(0, '/usr/local/lib/python2.7/site-packages')
    sys.path.insert(0, '/usr/local/lib/python2.7/site-packages/serial')
    print(sys.path)
    import serial

This made it able to find serial but then an import statement within the serial module failed to work.

    ['/usr/local/lib/python2.7/site-packages/serial', '/usr/local/lib/python2.7/site-packages', '/usr/local/lib/python2.7', '/Users/joepollard1/MyArduinoProject/Arduino-Makefile/bin', ..., '/Library/Frameworks/Python.framework/Versions/2.6/lib/python2.6/site-packages']
    Traceback (most recent call last):
      File "/Users/joepollard1/MyArduinoProject/Arduino-Makefile/bin/ard-reset-arduino", line 11, in <module>
        import serial
      File "/usr/local/lib/python2.7/site-packages/serial/__init__.py", line 11, in <module>
        import importlib
    ImportError: No module named importlib

So i'm at another dead end.

ladislas, 2017-03-12:
maybe your path is not set properly.

are you using homebrew? have you installed python with it?

String-bikinis, 2017-03-13:
I know I've used Homebrew before but it's been a long time and I don't remember which python versions I used it for.  So, as I understand what you're saying (and what seems to be the case), different python versions will be installed in different locations depending on what I use to install them. To be honest have no idea how to actually find those paths without searching with random file navigation.

ladislas, 2017-03-13:
can you run:

```
$ brew version
$ brew update && brew upgrade
$ brew list
$ which python
$ cat /etc/paths
```
and give me the output of `brew list`, `which python` and `cat /etc/paths`

String-bikinis, 2017-03-13:
'''
$ brew list
avr-binutils	avrdude		gmp		libmpc		mpfr		portaudio	qt		readline
avr-gcc		gdbm		libelf		libusb		openssl		pyqt		qt5		sip
avr-libc	git		libftdi0	libusb-compat	pkg-config	python		qwt		sqlite
$ which python
/Library/Frameworks/Python.framework/Versions/2.6/bin/python
$ cat /etc/paths
/usr/local/bin
/usr/bin
/bin
/usr/sbin
/sbin
'''

ladislas, 2017-03-13:
try `brew update && brew upgrade && brew remove python && brew install python`

String-bikinis, 2017-03-13:
```
$ brew update && brew upgrade && brew remove python && brew install python
Already up-to-date.
==> Upgrading 1 outdated package, with result:
homebrew/boneyard/pyqt 4.11.4
==> Upgrading homebrew/boneyard/pyqt 
==> Downloading https://downloads.sf.net/project/pyqt/PyQt4/PyQt-4.11.4/PyQt-mac
Already downloaded: /Users/joepollard1/Library/Caches/Homebrew/pyqt-4.11.4.tar.gz
==> python configure.py --confirm-license --bindir=/usr/local/Cellar/pyqt/4.11.4
Last 15 lines from /Users/joepollard1/Library/Logs/Homebrew/pyqt/01.python:
2017-03-13 12:28:11 -0600

python
configure.py
--confirm-license
--bindir=/usr/local/Cellar/pyqt/4.11.4/bin
--destdir=/usr/local/Cellar/pyqt/4.11.4/lib/python2.7/site-packages
--sipdir=/usr/local/Cellar/pyqt/4.11.4/share/sip

Error: /usr/local/opt/qt5/bin/qmake failed to create a makefile. Make sure you
have a working Qt qmake on your PATH or use the -q argument to explicitly
specify a working Qt qmake.
Determining the layout of your Qt installation...

Error: pyqt was moved to homebrew-boneyard because it has unfixable issues.
Please do not file any issues about this. Sorry!
```

ladislas, 2017-03-13:
can you `brew remove pyqt` and try again?

String-bikinis, 2017-03-13:
```
$ brew remove pyqt
Uninstalling /usr/local/Cellar/pyqt/4.11.1... (560 files, 17.7MB)
$ brew update && brew upgrade && brew remove python && brew install python
Already up-to-date.
Uninstalling /usr/local/Cellar/python/2.7.13... (3,526 files, 48MB)
python 2.7.9 is still installed.
Remove all versions with `brew uninstall --force python`.
==> Downloading https://homebrew.bintray.com/bottles/python-2.7.13.sierra.bottle
Already downloaded: /Users/joepollard1/Library/Caches/Homebrew/python-2.7.13.sierra.bottle.tar.gz
==> Pouring python-2.7.13.sierra.bottle.tar.gz
==> Using the sandbox
==> /usr/local/Cellar/python/2.7.13/bin/python -s setup.py --no-user-cfg install
==> /usr/local/Cellar/python/2.7.13/bin/python -s setup.py --no-user-cfg install
==> /usr/local/Cellar/python/2.7.13/bin/python -s setup.py --no-user-cfg install
==> Caveats
Pip and setuptools have been installed. To update them
  pip install --upgrade pip setuptools

You can install Python packages with
  pip install <package>

They will install into the site-package directory
  /usr/local/lib/python2.7/site-packages

See: http://docs.brew.sh/Homebrew-and-Python.html
==> Summary
🍺  /usr/local/Cellar/python/2.7.13: 3,526 files, 48MB
```

ladislas, 2017-03-13:
does it work now?

String-bikinis, 2017-03-13:
same error as before
```
mkdir -p /Users/joepollard1/MyArduinoProject/bin/FooProject/uno
/Library/Developer/CommandLineTools/usr/bin/make reset
/Users/joepollard1/MyArduinoProject/Arduino-Makefile/bin/ard-reset-arduino  /dev/cu.usbmodem1411
Traceback (most recent call last):
  File "/Users/joepollard1/MyArduinoProject/Arduino-Makefile/bin/ard-reset-arduino", line 4, in <module>
    import serial
ImportError: No module named serial
make[1]: *** [reset] Error 1
make: *** [upload] Error 2
```

ladislas, 2017-03-13:
wait wait wait! have you installed pySerial first?!

```
pip install pySerial
```

String-bikinis, 2017-03-13:
yes... but it only does it for version 3.5
```
$ pip install pySerial
Requirement already satisfied: pySerial in /Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages
```

ladislas, 2017-03-13:
mine says that 

```
→ pip install pySerial
Requirement already satisfied: pySerial in /usr/local/lib/python2.7/site-packages
```
that path is not the same... 

i would try uninstalling homebrew and starting again.

what is the output of `pip --version`, `pip2 --version`, `pip3 --version`?

ladislas, 2017-03-14:
so my guess is that for some reasons you've installed python from another source than homebrew and now it is in your path.

I don't know how to help...

closing this issue as it is not related to the project :(

closed 2017-03-14 by ladislas