ladislas/Bare-Arduino-Project #34: Builds fail for more complex user library structures

sherrellbc, 2017-01-28:
The provided example code has a user library structure as

> lib/FooLib/FooLib.h

Including this header works as expected in the current structure. However, if the intended header file name does not match the parent directory name then the build will fail with a "No such file or directory" error.

To reproduce, change lib/FooLib/FooLib.h to lib/FooLib/FooLib2.h and attempt to make.

ladislas, 2017-01-28:
yes it's a known issue and is due to the way Arduino-Makefile is made. I've tried working on it but never got the results I needed.

BUT, you can overcome that by having one header with the name of the folder and then including other headers in this one.

Just like here: https://github.com/weareleka/moti/tree/dev/lib/FreeIMU

I just have to include FreeIMU.h in my main.cpp

closed 2017-01-28 by ladislas

sherrellbc, 2017-01-28:
This is exactly as I did. I was hoping there would be a simpler solution.

sherrellbc, 2017-01-28:
This is exactly as I did. I was hoping there would be a simpler solution. It may be simpler to just pass the include (-I) option to gcc and require the user to qualify the origin of the header from in the include statement. Namely, 

#include <FooLib/FooLib.h>

ladislas, 2017-01-28:
yes that also works! 👍