ladislas/Bare-Arduino-Project #35: Not working in Ubuntu 14

Juanjbb, 2017-02-19:
First of all thank you for the great work!
I'm experiencing some problems setting the bare-bone project up in Ubuntu:

----------------------- 1st problem ------------------------
My avr-gcc version is 4.8.2, which in the Arduino.mk makes it try avr-gcc-ar

    ifeq ($(shell expr $(CC_VERNUM) '>' 482), 1)
        AR_NAME      = avr-gcc-ar
    else
        AR_NAME      = avr-ar
    endif

which produces the following error when trying to build:
'sorry - this program has been built without plugin support'

I changed $(shell expr $(CC_VERNUM) '>' 482 for
$(shell expr $(CC_VERNUM) '>' 483

to make it use avr-gcc, which compiles without errors

----------------------- 2nd problem ------------------------

Same version problem for avr dude

            ifndef AVRDUDE
                ifeq ($(shell expr $(ARDUINO_VERSION) '>' 157), 1)
                    # 1.5.8 has different location than all prior versions!
                    AVRDUDE = $(AVR_TOOLS_DIR)/bin/avrdude
                else
                    AVRDUDE = $(AVR_TOOLS_DIR)/../avrdude
                endif
            endif

            ifndef AVRDUDE_CONF
                ifeq ($(shell expr $(ARDUINO_VERSION) '>' 157), 1)
                    AVRDUDE_CONF = $(AVR_TOOLS_DIR)/etc/avrdude.conf
                else
                    AVRDUDE_CONF = $(AVR_TOOLS_DIR)/../avrdude.conf
                endif
            endif

I have version 105 but the paths are just the other, so I changed 
($(shell expr $(ARDUINO_VERSION) '>' 157) 
for
($(shell expr $(ARDUINO_VERSION) '>' 100)

-----------------------------

I know is a dirty work around, and that's why I reported the issue so you guys can think for a better fix! Thank you again for the great work

Juanjbb, 2017-02-19:
Ok, the second problem I think it was my fault. I installed arduino long time ago and I may had changed the directories. The first problem though, no clue why I get the 'sorry - this program has been built without plugin support' error

ladislas, 2017-03-12:
thanks for the feedback.

this is more of an upstream bug from what I see. I'll look into it and report it if it's the case.

closed 2017-03-12 by ladislas