ladislas/Bare-Arduino-Project #37: vim YouCompleteMe still not working with .ino files

UsoHolger, 2017-03-11:
I placed .ycm_extra_conf.py in the project directory but YCM still does not work at all. For example: When I type `test.` where test is the name of an object of a test class I do not get any suggestions for method names. When I try to force autocompletion I get
```E764: Option 'omnifunc' is not set```

When I try to use the go to definition feature the error is
``` ValueError: No semantic completer exists for filetypes: [u'arduino']```

ladislas, 2017-03-11:
hmm...

I'm not sure it works with `.ino` files, and this project doesn't support `.ino` files either.

can you post your `.ycm_extra_conf` somewhere, so I can take a look?

UsoHolger, 2017-03-12:
Thanks for the reply. I am using [this](https://gist.github.com/ajford/f551b2b6fd4d6b6e1ef2) version, which is based on yours. But it seems like it is not even loaded. When I deliberately add a syntax error I do not get any notification nor did YCM ever ask me to load the config. I am using a global config for c++11 which I placed in my ~/.vim folder. It is loaded in my .vimrc with 
```
let g:ycm_global_ycm_extra_conf = '~/.vim/.ycm_global_ycm_extra_conf.py'
```
but I thought that is only being used as a fallback so local configs should still be loaded.

ladislas, 2017-03-12:
yep, the issue is the one I thought, it doesn't recognize the arduino file type.

setting filetype to cpp should not stop highlighting of functions. have you tried https://github.com/sheerun/vim-polyglot

either way, it is outside the scope of this project :)

I'm closing the issue as it has been resolved.

closed 2017-03-12 by ladislas