ladislas/Bare-Arduino-Project #38: How to properly use libraries and OOP?

UsoHolger, 2017-03-13:
This is a question. Not really sure whether to ask this here or in [Arduino-Makefile](https://github.com/sudar/Arduino-Makefile).

I personally don't really care about compatability to the arduino IDE I just want to use theire libraries. With this repo and the makefile I kind of have what I wanted except for the way libraries are compiled, which I do not understand fully. 

By reading the code I found that the folder names of libraries have to be the same as the .h and .cpp file and there can only be one .h and one .cpp file per library is that correct?

So I like clean OOP with files for each class, I know one should not go to heavily object oriented for hardware programming but from what I have read it is okay as long as one does not use polymorphy etc. since the compiler can't optimize that. And not to use the heap to much. So what I do is create one "library" for each class and create objects in the main file as global variables, is that the way to do it?

I tried to play around with [StandardCplusplus](https://github.com/maniacbug/StandardCplusplus) but could not make it work in this framework even though I tried to edit the regex in the Arduino-Makefile to account for the missing .h. I am not saying I want to use the bloated container classes but e.g. functional would be nice to have. So just as an example: How would I make such a library with multiple .cpp and header files work, is that even possible without changing the entire makefile?

ladislas, 2017-03-13:
> By reading the code I found that the folder names of libraries have to be the same as the .h and .cpp file and there can only be one .h and one .cpp file per library is that correct?

no you can have as many `.h` and `.cpp` as you'd like but they have to be referenced into the main `MyLibrary.h` file inside the `MyLibrary` folder.

Just like here: https://github.com/weareleka/moti/tree/dev/lib/FreeIMU

> So I like clean OOP with files for each class, I know one should not go to heavily object oriented for hardware programming but from what I have read it is okay as long as one does not use polymorphy etc. since the compiler can't optimize that.

yep -- but don't worry, if something is not supported, it will just not compile. But as main rule in embedded development, try to avoid creating objects while the application runs, try to do it just as the beginning.

> I tried to play around with StandardCplusplus but could not make it work in this framework

try adding putting all the libs into a `StandardCpluplus` folder, then create a `StandardCplusplus.h` file and `#include` all the libraries you need, then `#include "StandardCplusplus.h"`  in your `main.cpp`

UsoHolger, 2017-03-13:
Thanks a lot that really helped me. Now I have the perfect enviroment to get started with some bigger projects. Great work!

closed 2017-03-13 by UsoHolger