ladislas/Bare-Arduino-Project #39: No rule to make target `build-mega-atmega2560/src.eep', needed by `all'

rpbeltran, 2017-03-17:
I'm trying to create a new project, but I have been unable to get any makes to work

My makefile looks like:
```
PROJECT_DIR       := /Users/ryanbeltran/Documents/Hardware\ Control/hardware\ controller/src
ARDMK_DIR         = $(PROJECT_DIR)/../Arduino-Makefile
ARDUINO_DIR       = /Applications/Arduino.app/Contents/Java
AVR_TOOLS_DIR     = /usr/local
AVRDDUDE          = /usr/local/bin/avrdude
BOARD_TAG         = mega
BOARD_SUB         = atmega2560
### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 9600
### CFLAGS_STD
CFLAGS_STD        = -std=gnu11
CXXFLAGS_STD      = -std=gnu++11
CXXFLAGS          = -pedantic -Wall -Wextra -fdiagnostics-color
### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT      = /dev/cu.usbmodem*
### path to Arduino.mk, inside the ARDMK_DIR, don't touch.
include $(ARDMK_DIR)/Arduino.mk
```

Everytime I build it prints out all the configuration data and then the error message:
`make: *** No rule to make target build-mega-atmega2560/src.eep', needed by all'.  Stop.`

I tried running `cat /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt` and found the line `mega.menu.cpu.atmega2560=ATmega2560 (Mega 2560)` which makes me think I should be using the correct values.

Any help would be greatly appreciated. Let me know if any more information would be of help.

ladislas, 2017-03-17:
I've never had this issue before.

can you point me to the repository where the code you're trying to compile is so that i can test it myself?

rpbeltran, 2017-03-17:
I uploaded my code to https://github.com/rpbeltran/hardware-controller I also tried to make it with a UNO as the target, but no luck there either, just got the same error but changed to `build-uno-atmega328/src.eep' Thank you for the help.

ladislas, 2017-03-17:
oh i know! it's because inside of `src` you need another folder with you "project" and the `main.cpp`, `Makefile`, etc. just like here: https://github.com/weareleka/moti/tree/dev/src

each folder is a different project with it's own `main.cpp` and they all compile.

I guess i could change the behavior, it should not be that hard.

the folder `multiplexer` and so on, must be in `lib`

let me know if that works

rpbeltran, 2017-03-17:
That fixed everything, thank you. I also realized that you can not put your project inside of a path with any spaces in it, but I think that is a problem with make, not anything here. 

Thank you for the help, and also for making this. It's an awesome project.

closed 2017-03-17 by rpbeltran

ladislas, 2017-03-17:
my pleasure :)