ladislas/Bare-Arduino-Project #43: Wrong instructions?

pip010, 2017-04-23:
If I follow the instruction.md correctly you end up with a project root folder with Makefile and /src /lib /bin
subfolders. However running make gives this:

-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/share/arduino 
- [USER]               ARDMK_DIR = /home/ppetrov/Documents/arduiono_example/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 105 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK =  
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /home/ppetrov/Documents/arduiono_example/lib 
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = atmega328P 
- [USER]               BOARD_TAG = uno 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [USER]               OBJDIR = /home/ppetrov/Documents/arduiono_example/bin/arduiono_example/uno 
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
/home/ppetrov/Documents/arduiono_example/Arduino-Makefile/Arduino.mk:780: *** At least one source file (*.ino, *.pde, *.cpp, *c, *cc, *.S) is needed.  Stop.

which makes sense since no source file in project root, did I miss something?

ladislas, 2017-04-23:
you have to run `make` in `./src/myProject`

ladislas, 2017-04-23:
like from here https://github.com/ladislas/Bare-Arduino-Project/tree/master/src/BarProject

pip010, 2017-04-23:
Thanks, that did it. But running then 'make upload' from same dir produces:
/usr/bin/avrdude -q -V -p atmega328p -C /usr/../avrdude.conf -D -c arduino -b 115200 -P /dev/ttyUSB0 \
		-U flash:w:/home/ppetrov/Documents/arduiono_example/bin/FooProject/uno/FooProject.hex:i
avrdude: can't open config file "/usr/../avrdude.conf": No such file or directory

I fixed it manually, by pointing to /etc/avrdude.conf
still my board times out when flashed, not sure whether I need different .conf file for this specific board 'BQ ZUM Core'.

closed 2017-04-23 by pip010

ladislas, 2017-04-23:
yeah, i've seen this error with avrdude before. sometimes it happens on linux and people need to fix it by hand.

I don't use linux so I can't help, but you might find something here: https://github.com/sudar/Arduino-Makefile/issues?utf8=%E2%9C%93&q=avrdude%20conf

JBaczuk, 2018-07-29:
Same issue on Ubuntu 16.04. Just add this to the Makefile:

`AVRDUDE_CONF = /etc/avrdude.conf`