ladislas/Bare-Arduino-Project #46: makefile not working with tinyGPS and tinyGPSPlus

kamze, 2017-06-01:
I am trying to compile an example of tinyGPSPlus (same goes tinyGPS) for it works fine when i compile it using arduino IDE but when i use make upload there is no error but i don't receive any GPS data.
Please note that i have all ready uploaded functional programs using make upload.
Could it because of the baudrate mismatch talked about here :https://github.com/ladislas/Bare-Arduino-Project/issues/45

ladislas, 2017-06-01:
can you try printing something else right before the GPS data. it will help us figure out if it the printing that doesn't work or just the data.

kamze, 2017-06-01:
so the code is : 
#include <TinyGPSPlus.h>
#include <SoftwareSerial.h>
/*
   This sample sketch demonstrates the normal use of a TinyGPS++ (TinyGPSPlus) object.
   It requires the use of SoftwareSerial, and assumes that you have a
   4800-baud serial GPS device hooked up on pins 4(rx) and 3(tx).
*/
static const int RXPin = 3, TXPin = 2;
static const uint32_t GPSBaud = 4800;

// The TinyGPS++ object
TinyGPSPlus gps;

// The serial connection to the GPS device
SoftwareSerial ss(RXPin, TXPin);

void setup()
{
  ss.begin(GPSBaud);
  Serial.begin(9600);
}

void loop()
{
  Serial.println("-----------Test---------- ");

  while (ss.available() > 0)
    if (gps.encode(ss.read()))
      if (gps.location.isValid())
  {
    Serial.print("Location: ");
    Serial.print(gps.location.lat(), 6);
    Serial.print(",");
    Serial.println(gps.location.lng(), 6);
  }

}
using arduino's IDE with 9600 baud i get this : 
----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
Location: 48.299602,4.064558
-----------Test---------- 
-----------Test---------- 
Location: 48.299602,4.064558
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
Location: 48.299602,4.064558
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
Location: 48.299602,4.064558
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
Location: 48.299602,4.064558

but using make upload on 4800 (i didn't change the code at all if put 9600 baud i get these weird caracters) i get this : 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test---------- 
-----------Test----------

kamze, 2017-06-01:
i set GPSBaud to 9600 and it worked but i dont know why

ladislas, 2017-06-01:
the baudrate in the makefile is only to use `make monitor` and must be set to the same value as `Serial.begin(9600);`

The GPS baudrate might be something else, I don't know what it is.

closing this issue as it is out side of the scope of this repo

closed 2017-06-01 by ladislas

kamze, 2017-06-01:
ok thank you