ladislas/Bare-Arduino-Project #50: ib-detection gets stuck searching for dependencies

rpbeltran, 2018-01-10:
I went to compile my code to find that it was never finishing (after running for about ~45 minutes). When I would interrupt the process it would display a traceback to `Arduino-Makefile/bin/lib-detection` somewhere around line 70 ( `currentFile = open(os.path.join(USER_LIB_PATH, lib, lib + ".h"))` )

For some reason, it was getting stuck calculating `SKETCH_LIBS_DEPS`. As a workaround, I specified every library and every library dependency directly in the `ARDUINO_LIBS` variable of the makefile, commented out that section of lib-detection, and replaced it with `SKETCH_LIBS_DEPS = []`. This approach seems to be working for now.

[image]

My project is somewhat large with 15 libraries specified in my USER_LIB_PATH. Each individual library is fairly simple, with a handful of includes and an average size of ~200 - 300 lines, with a couple libraries being substantially larger. Some libraries include other libraries. Hardcoding dependencies in the makefile without commenting out lib-detection did not help.

Unfortunately, I cannot upload my project as it's for work and parts of it are confidential, but would be happy to try and provide any specific information that might be helpful.

Thanks!

CURRENT_OS = MAC 
ARDUINO_VERSION = 165-r 
ARDMK_VERSION = 1.5

ladislas, 2018-01-10:
it might have to do with recursive library inclusion or deeper nested levels.

in the current state of things, there's not much that can be done unless you want to get your hand dirty and try to find a solution.

I'm not using the bare arduino project that much these days and I'm slowly moving to platformio which offers much more customization options. they also provide you with auto lib detection that really works.

rpbeltran, 2018-01-10:
Thanks, if I get the chance I'll look into it further and see if I can find anything. If I come up with a real fix I'll update it here and put in the PR.

ladislas, 2018-01-10:
👍🏻

-- 
---

This e-mail and any attached files may contain confidential information 
which is intended exclusively for the recipient. If you are not the 
specified recipient, you should delete this e-mail from your system 
immediately, and inform the sender without delay. Reading, distributing, 
copying, or any other unauthorized use of this e-mail and its attached 
files is strictly forbidden. Thank you for your comprehension.

Cet e-mail et ses éventuelles pièces jointes peuvent contenir des 
informations confidentielles, lesquelles sont envoyées à l’intention 
exclusive de son destinataire. Si vous n’êtes pas le destinataire auquel 
est envoyé cet e-mail, vous devez immédiatement l’effacer de votre système 
et avertir son expéditeur. Toute lecture, distribution, copie ou autre 
utilisation de cet e-mail et de ses éventuelles pièces jointes est 
strictement interdite. Merci.

ladislas, 2018-08-22:
closing for now, feel free to reopen.

closed 2018-08-22 by ladislas