ladislas/Bare-Arduino-Project #51: Work with ArduinoCore-avr: troubles

ktrace, 2018-05-12:
Good day!
I try to setup Bare-Arduino-Project without Arduino at all, but with ArduinoCore-avr (https://github.com/arduino/ArduinoCore-avr).
Step to reproduce:

```
cd work
git clone https://github.com/arduino/ArduinoCore-avr.git
git clone https://github.com/ladislas/Bare-Arduino-Project.git
cd Bare-Arduino-Project && git submodule update --init --recursive
cd src/FooProject
cp ../../Makefile-Linux.mk Makefile
```

Edit Makefile same way:

```
diff Makefile ../../Makefile-Linux.mk
20,21c20,21
< ARDUINO_DIR       = ~/work/ArduinoCore-avr
< ALTERNATE_CORE_PATH = ~/work/ArduinoCore-avr
---
> ARDUINO_DIR       = /usr/share/arduino
>
```
Now after make:
```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX
- [USER]               ARDUINO_DIR = ~/work/ArduinoCore-avr
- [USER]               ARDMK_DIR = /home/user/work/Bare-Arduino-Project/Arduino-Makefile
- [DEFAULT]            ARDUINO_VERSION = 100
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            ARDMK_VENDOR = arduino
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/user/.arduino/preferences.txt
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/user/sketchbook (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr
- [COMPUTED]           ARDUINO_LIB_PATH = ~/work/ArduinoCore-avr/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE_PATH = ~/work/ArduinoCore-avr
- [COMPUTED]           ARDUINO_VAR_PATH = ~/work/ArduinoCore-avr/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = ~/work/ArduinoCore-avr/boards.txt (from ALTERNATE_CORE_PATH)
- [USER]               USER_LIB_PATH = /home/user/work/Bare-Arduino-Project/lib
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_SUB = atmega2560
- [USER]               BOARD_TAG = mega
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = mega (from build.variant)
- [USER]               OBJDIR = /home/user/work/Bare-Arduino-Project/bin/FooProject/mega
- [COMPUTED]           ARDUINO_CORE_PATH = ~/work/ArduinoCore-avr/cores/arduino (from ALTERNATE_CORE_PATH, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
-  [USER]              MONITOR_BAUDRATE = 115200
/home/user/work/Bare-Arduino-Project/Arduino-Makefile/Arduino.mk:974: *** The following libraries specified in ARDUINO_LIBS could not be found (searched USER_LIB_PATH and ARDUINO_LIB_PATH): Wire. Stop.
```
Here two errors:
1. - [COMPUTED]           ARDUINO_CORE_PATH = ~/work/ArduinoCore-avr/cores/arduino (from ALTERNATE_CORE_PATH, BOARD_TAG and boards.txt) - not exist
2. Wire not found. Wire lays on 
```
$ ls ~/work/ArduinoCore-avr/libraries
EEPROM  HID  SoftwareSerial  SPI  Wire
$ ls ~/work/ArduinoCore-avr/libraries/Wire/
examples  keywords.txt  library.properties  src
$ ls -l ~/work/ArduinoCore-avr/libraries/Wire/src
total 20
-rw-r--r-- 1 user user 8591 May 12 00:22 Wire.cpp
-rw-r--r-- 1 user user 2682 May 12 00:22 Wire.h
drwxr-xr-x 2 user user 4096 May 12 00:22 utility

```
Need tips to fix it.

ladislas, 2018-08-22:
it doesn't seem possible right now as the arduino app packages a lot of different things that are hard wired with the makefile.
it might change in the future but for the moment you need to download the arduino app...

ladislas, 2018-08-22:
closing for now, feel free to reopen if something changes in the future.

closed 2018-08-22 by ladislas