ladislas/Bare-Arduino-Project #55: C

CedricMi, 2019-04-11:
Hello,
I'm trying this project under an Ubuntu 18.04.2 WSL of Windows 10 fully patched 1809/17763.437.

1. In Step 5 of INSTALL.MD
make fails because avrdude.conf is not found.

My avrdude.conf is under 
/usr/share/Arduino/hardware/tools/arvdude.conf

I had to change Arduino.MK lines 576+ to
   # ensure we can still find avrdude.conf
    ifndef AVRDUDE_CONF
	#CM
		AVRDUDE_CONF = /usr/share/arduino/hardware/tools/avrdude.conf
#       ifeq ($(shell expr $(ARDUINO_VERSION) '>' 157), 1)
#            AVRDUDE_CONF = $(AVR_TOOLS_DIR)/etc/avrdude.conf
#        else
#            AVRDUDE_CONF = $(AVR_TOOLS_DIR)/../avrdude.conf
#        endif
    endif

2. In Step 5 of INSTALL.MD
make upload fails
cedric@CEDRIC-DSK1:/mnt/f/dev/GitHub/bat02/src/FooProject$ make upload
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX
- [USER]               ARDUINO_DIR = /usr/share/arduino
- [USER]               ARDMK_DIR = /mnt/f/dev/GitHub/bat02/Arduino-Makefile
- [AUTODETECTED]       ARDUINO_VERSION = 105
- [DEFAULT]            ARCHITECTURE =
- [DEFAULT]            ARDMK_VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK =
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
- [USER]               AVR_TOOLS_DIR = /usr
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /mnt/f/dev/GitHub/bat02/lib
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_SUB = atmega2560
- [USER]               BOARD_TAG = mega
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = mega (from build.variant)
- [USER]               OBJDIR = /mnt/f/dev/GitHub/bat02/build/FooProject/mega
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-                      No .pde or .ino files found. If you are compiling .c or .cpp files then you need to explicitly include Arduino header files
-  [USER]              MONITOR_BAUDRATE = 115200
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [USER]               CFLAGS_STD = -std=gnu11
- [USER]               CXXFLAGS_STD = -std=gnu++17
- [COMPUTED]           DEVICE_PATH = /dev/ttyS10 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      SKETCH_LIBS =
- [USER]                 FooLib
-
-                      SYSTEM_LIBS =
- [SYSTEM]               Wire
-
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6
- [COMPUTED]           CC_VERSION = 5.4.0 (avr-gcc)
-------------------------
mkdir -p /mnt/f/dev/GitHub/bat02/build/FooProject/mega
make reset
make[1]: Entering directory '/mnt/f/dev/GitHub/bat02/src/FooProject'
/mnt/f/dev/GitHub/bat02/Arduino-Makefile/bin/ard-reset-arduino  /dev/ttyS10
make[1]: Leaving directory '/mnt/f/dev/GitHub/bat02/src/FooProject'
make do_upload
make[1]: Entering directory '/mnt/f/dev/GitHub/bat02/src/FooProject'
/usr/bin/avrdude -q -V -p atmega1280 -C /usr/share/arduino/hardware/tools/avrdude.conf -D -c arduino -b 57600 -P /dev/ttyS10 \
                -U flash:w:/mnt/f/dev/GitHub/bat02/build/FooProject/mega/FooProject.hex:i
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

/mnt/f/dev/GitHub/bat02/Arduino-Makefile/Arduino.mk:1725: recipe for target 'do_upload' failed
make[1]: *** [do_upload] Error 1
make[1]: Leaving directory '/mnt/f/dev/GitHub/bat02/src/FooProject'
/mnt/f/dev/GitHub/bat02/Arduino-Makefile/Arduino.mk:1712: recipe for target 'upload' failed
make: *** [upload] Error 2
cedric@CEDRIC-DSK1:/mnt/f/dev/GitHub/bat02/src/FooProject$

Note: The ATMEGA 2560 board is working. Directly under Windows with Arduino IDE, uploading sketches does work.
Any ideas?

ladislas, 2019-04-11:
You're using a very old version or the Arduino IDE, 1.0.5.

Update Arduino, try again and let me know.

CedricMi, 2019-04-11:
Ok didn't know one couldn't count on Ubuntu's repos to be up to date for Arduino. May be something you should mention it in INSTALL.MD

I installed the Arduino IDE from Arduino.cc
make upload still couldn't find avrdude.conf, with original or with my mod, and refused to upload

So I uninstalled then reinstalled gcc-avr binutils avr-libc avrdude
Seems even worse, now I get this during make
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
/mnt/f/dev/GitHub/bat02/Arduino-Makefile/Arduino.mk:437: *** Currently BOARDS_TXT='/usr/share/arduino/hardware/arduino//boards.txt', which is not an existing file or an invalid filename..  Stop.

Pretty strange, since COMPUTED is right, but Board in error message is wrong (double slash):
BOARDS_TXT='/usr/share/arduino/hardware/arduino**//**boards.txt', which is not an existing file or an invalid filename..  Stop.

Any idea? otherwise I'll start from a blank linux install.

ladislas, 2019-04-12:
Hmm, board.txt is in the Arduino installation folder. Can you check you can find it somewhere here `/usr/share/arduino/hardware/arduino`?

The double slash should not be an issue.

Bare-Arduino-Project uses https://github.com/sudar/Arduino-Makefile/, try using it "vanilla" and see what happens.

ladislas, 2019-04-12:
I've also never used Ubuntu WSL so I'm not even sure it is supposed to work.

You should try it using VMWare or equivalent. Or just make a dual boot and boot on linux.

CedricMi, 2019-04-12:
Ok, after tons of problems, spending way too much time fixing them, and an additional parallel install to a true ubuntu virtual machine, my conclusion is that the install file arduino-1.8.9-linux64.tar.xz downloaded from Arduino.cc does somehow not install correctly on Ubuntu & debian-like distributions.
So instead of defining ARDUINO_DIR as /Arduino or /.arduino15, one has to point to the extracted archive, such as:
/home/MyuserName/Downloads/Arduino-1.8.9

Having done that, make upload still complains about not finding avrdude.conf, this time under /usr/etc/avrdude.conf
I did it, then everything is fine.

TLDR:
1. Latest Arduino IDE 1.8.9 does not install correctly on Debian distributions, or at least Ubuntu & Pengwin. I'll try to find where to feedback.
2. Bare-Arduino-Projet works under WSL, which is fantastic

Question: The way it is done with Bare Arduino Project, own Arduino projects must be under a subdirectory.
Ideally, Arduino projets are separate, and somehow reference Bare Arduino Project for building and uploading:

_ projects
    -arduinop1
    -arduinop2
    -...
    -MyBareArduinoProjectClone

How would you go about that ?

Cheers
Cedric

ladislas, 2019-04-12:
Good to know you fixed it.

I myself put the different projects in `src/` just like here: https://github.com/leka/moti/tree/dev/src

Or I clone Bare-Arduino-Project for every new projects.

Checkout the documentation of https://github.com/sudar/Arduino-Makefile/ as they explain how to have Arduino-Makefile in only one place and configure your local makefiles.

closed 2019-04-12 by ladislas