ladislas/Bare-Arduino-Project #56: How can I integrate unittests?

dolfandringa, 2019-07-24:
I love you made this project, it was kind of what I was thinking of myself using the Arduino-Makefile project as well, but you streamline it a lot. I was also planning to do CI and unittesting, but although the test folder is there, you don't mention any testing framework, library, etc  that you use for it. I was planning to use [arduino-ci](https://github.com/ianfixes/arduino_ci) because it provides a lot of hardware mocking functionality so it allows me to focus my unittests on my actual code. Do you have any experience integrating that into your Bare-Arduino-Project? Or how else are you doing testing in your projects? Any pointers on this would be greatly appreciated, and I'd be happy to contribute docs on this.

dolfandringa, 2019-07-25:
Some more investigation, I think I can just run the runtests.sh script as the test discovery, instead of using the arduino_ci supplied one. I just need to make sure that when running tests, instead of including the regular Arduino.h, I include the arduino_ci supplied one. Are there any quick pointers on how to achieve that without directly modifying the Arduino-Makefile scripts?

dolfandringa, 2019-07-26:
Ok, I am getting further with this. I indeed don't want to use the arduino_ci test runner. Just the Arduino library for mocking and testing. I managed to extract the bare minimum compile command that their test runner uses for my unittests, which is this:
```
g++ -std=c++0x -o /home/me/Bare-Arduino-Project/unittest_test_sample.cpp.bin \
    -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__AVR_ATmega328P__ \
    -I/usr/share/gems/gems/arduino_ci-0.2.0/cpp/arduino \
    -I/usr/share/gems/gems/arduino_ci-0.2.0/cpp/unittest \
    -I/home/me/Bare-Arduino-Project/lib/testlib \
    /usr/share/gems/gems/arduino_ci-0.2.0/cpp/arduino/Arduino.cpp \
    /usr/share/gems/gems/arduino_ci-0.2.0/cpp/arduino/Godmode.cpp \
    /usr/share/gems/gems/arduino_ci-0.2.0/cpp/arduino/stdlib.cpp \
    /usr/share/gems/gems/arduino_ci-0.2.0/cpp/unittest/ArduinoUnitTests.cpp \
    /home/me/Bare-Arduino-Project/lib/testlib/testlib.cpp \
    /home/me/Bare-Arduino-Project/test/testlib/test_sample.cpp
```

I guess I need to make sure my Makefile-CI only includes the Arduino sources from arduino-ci and doesn't include the avr-libc libraries either, and then prepends the Arduino.cpp and Godmode.cpp  files before my testlib.cpp and test_sample.cpp. Any ideas if this is possible with only modifying `scripts/*` and the `Makefile-*` files from `Bare-Arduino-Projects`?

dolfandringa, 2019-07-26:
damn, I just realize I overlooked that arduino_ci uses native gcc not avr-gcc, so the arduino_ci libraries will probably never work with avr-gcc and therefore Bare-Arduino/Arduino-Makefile, right?

dolfandringa, 2019-07-27:
I have managed to include arduino_ci after I contributed a feature to it that resolved conflicts between Bare-Arduino-Project and arduino_ci. I forked this repos and have the combined repos on https://github.com/dolfandringa/Bare-Arduino-Project for those interested. I'd rather just add a good mocking library for arduino to Bare-Arduino-Project rather than combining the 2, but this works for me for now.

closed 2019-07-27 by dolfandringa

ladislas, 2019-08-27:
@dolfandringa thanks for the input and sorry for the delayed answer. I don't use any framework for the moment but thanks to you I've discovered Arduino CI and it seems to be very powerful. The fact that it uses gcc allow the tests to be run on your own computer without hardware and forces you to architecture your app in a smart way (as I said in #57).

One thing you could try is to symlink all the gcc bins and prefix them with `avr-` then set `AVR_TOOLS_DIR` in your Makefile to use it.

I haven't tried it so I can't say if it work, but I'll try to give it a try when I get time.