ladislas/Bare-Arduino-Project #57: Mocking SPI and hardware pins

dolfandringa, 2019-07-26:
Related to #56: Is there a library or are there any examples of writing unittests that Mock some of the hardware arduino functions? I would for instance like to mock the digitalWrite/digitalRead/analogRead/pinMode functions, and also the SPI library, so I can write tests for code that depends on them, without actually needing an emulator or physical hardware connected. I really like the arduino_ci project for that specific part, but they have their own test runner in Ruby and use the native gcc instead of avr-gcc. Is something similar possible using avr-gcc?

ladislas, 2019-08-27:
If you want to run the test without actual hardware, you won't be able to use avr-gcc. You'll need to compile using gcc to run the program for your computer (either macOS, Linux our Windows). 

arduino-ci does provide SPI functions. I have never used it but the documentation seems pretty explicit: 

> https://github.com/ianfixes/arduino_ci/blob/master/REFERENCE.md#spi

The goal of arduino-ci is to allow you to write test that will run for real when compiled with avr-gcc and will run for tests when compiled with gcc.

Your goal as the developper is to structure your code so that it is testable. Don't be afraid to use C++ and OOP to make that easier.

ladislas, 2020-02-04:
closing as there's not activity -- feel free to reopen if needed

closed 2020-02-04 by ladislas