ladislas/Bare-Arduino-Project #60: Support for 3rd Party Boards

DouglasUrner, 2020-04-24:
Is it possible to build projects for 3rd party boards such as the Adafruit Trinket (the ATTiny85 version)?

I have the Adafruit board support installed in the Arduino IDE, but they aren't recognized by the Makefile, for example:

```
make show_boards
```

Doesn't show the Adafruit boards. Is there a way to pick up the 3rd party board support from the IDE (i.e., see the same boards that the IDE sees)? If there isn't, would it work to add entries to **boards.txt**?

ladislas, 2020-04-24:
I don't know as I've never tried :(

You should check with https://github.com/sudar/Arduino-Makefile/ -- there might be some information there.

my guess is that in the worst case you should be able to define all the variables needed by the makefile that are usually found in board.txt. If you don't change your board often, it's a one time work and you're good to go.

ladislas, 2020-04-27:
@DouglasUrner any news?

ladislas, 2020-05-04:
@DouglasUrner closing this issue as it is not directly related to BAP. Reopen if needed.

closed 2020-05-04 by ladislas