ladislas/Bare-Arduino-Project #61: python 2,3 confusion

MalcolmBoura, 2020-09-22:
On my system,  Kubuntu 18.04, python defaults to python2 causing lib-detection to fail. Changing the shebang line to invoke python3 instead of python fixed the problem.

...,lib-detection", line 23
  print(lib, end = '  ')

The print as a function and end = construct was introduced in python3.

ladislas, 2020-09-22:
Thanks for reporting @MalcolmBoura! If you want to provide a PR, please do, I dont have much time to work on that.

As Python 2 is deprecated, the default python version should be Python 3. It's the case on macOS.

MalcolmBoura, 2020-09-24:
Thanks.

I have confirmed that python2.7 is the default for Ubuntu 18.04. That is an LTS release so it will be maintained until 2023. Changing the default python is likely to break a lot of things.

I will put in a PR but I will need to revise my git skills first!

ladislas, 2021-01-18:
@MalcolmBoura I'll close for now, feel free to reopen if needed.

closed 2021-01-18 by ladislas