ladislas/Bare-Arduino-Project #62: Sub-optimal diagnostics and documentation of BOARD_SUB

MalcolmBoura, 2020-09-22:
The documentation would benefit from an explanation of how to identify BOARD_TAG and BOARD_SUB from the boards.txt file.

Failure to find BOARD_SUB (and also I expect BOARD_TAG) should be reported as an error. avr-g++ complaining about lack of a value for -mmcu is rather obscure.

Having said that, thanks for the excellent project.

ladislas, 2020-09-22:
This is rather related to Arduino-Makefile, you can find the info here:

- https://github.com/sudar/Arduino-Makefile#useful-variables
- https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#board_tag
- https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#board_sub

MalcolmBoura, 2020-09-24:
Thanks for the URLs. I agree.

closed 2021-01-18 by ladislas