ladislas/Bare-Arduino-Project #63: No such file or directory Error when setting up a new sub project

joesan, 2020-12-31:
I tried the install instructions and created my first sub project, but when I wanted to run the make command, I got the following error:

```
joesan@joesan-InfinityBook-S-14-v5:~/Projects/Private/hardware-projects/navo-hardware-libs/src/MotorControlWithEncoder$ make
Makefile:70: /home/joesan/Projects/Private/hardware-projects/navo-hardware-libs/src/Arduino-Makefile/Arduino.mk: No such file or directory
make: *** No rule to make target '/home/joesan/Projects/Private/hardware-projects/navo-hardware-libs/src/Arduino-Makefile/Arduino.mk'.  Stop.
```

Why is it looking for Arduino.mk file? From where should I source this one?

ladislas, 2020-12-31:
You must `git submodule update --init --recursive` from your clone directory.

ladislas, 2020-12-31:
if you have the repository on github it would help me help you. I'm surprised by the path `../src/Arduino-Makefile/Arduino.mk`

how did you clone the project?

joesan, 2021-01-01:
I think I kind of fixed it by installing the https://github.com/sudar/Arduino-Makefile and then modifying the ARDMK_DIR in my Makefile point to 
        `  /usr/share/arduino/`

You mean to say if I run the `git submodule update --init --recursive`, I do not have to install the Arduino-Makefile explicitly?

ladislas, 2021-01-01:
Yes, the submodule thingy pulls a fork of sudar's arduino-makefile with additional capabilities. 

It's in the install documentation ;)

ladislas, 2021-01-18:
@joesan any news on this? I'll close for now, feel free to reopen if needed.

closed 2021-01-18 by ladislas