ladislas/Bare-Arduino-Project #65: Not working with newest Arduino - fixes

JDuchniewicz, 2021-04-21:
Not opening a PR because I noticed that you have your own fork of Arduino-Makefile project and maintain it.

Took me some time to figure it out but now it works.

First of all, with new SAMD devices, this will fail to compile and it requires changes in the README (might open a PR just for that). 
For the SAMD boards, there is no need for `BOARD_SUB`, everything is stored in `BOARD_TAG`. 

With the newest commits from the [upstream repository](https://github.com/sudar/Arduino-Makefile), the example project will still not build due to missing LTO plugins. 

Suggested change is as follows:
```
diff --git a/Arduino.mk b/Arduino.mk
index 5a04506..9258f0f 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -1073,6 +1073,9 @@ CAT     = cat
 ECHO    = printf
 MKDIR   = mkdir -p

+# SPECIAL WORKAROUND FOR LIBRARIES
+LTO_PLUGIN_DIR = $(HOME)/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/lib/gcc/arm-none-eabi/7.2.1/liblto_plugin.so
+
 # recursive wildcard function, call with params:
 #  - start directory (finished with /) or empty string for current dir
 #  - glob pattern
@@ -1698,7 +1701,7 @@ else
 endif

 $(CORE_LIB):   $(CORE_OBJS) $(LIB_OBJS) $(PLATFORM_LIB_OBJS) $(USER_LIB_OBJS)
-               $(AR) rcs $@ $(CORE_OBJS) $(LIB_OBJS) $(PLATFORM_LIB_OBJS) $(USER_LIB_OBJS)
+               $(AR) rcs $@ $(CORE_OBJS) $(LIB_OBJS) $(PLATFORM_LIB_OBJS) $(USER_LIB_OBJS) --plugin $(LTO_PLUGIN_DIR)

 error_on_caterina:
                $(ERROR_ON_CATERINA)
```

I did not find a way to replace the `arm-none-eabi-gcc-ar` with a C++ one so this requires the LTO Plugin to be included. Feel free to alter it. Also I opened a [PR](https://github.com/sudar/Arduino-Makefile/pull/659) upstream to fix zeroing the board before flashing.

ladislas, 2021-04-21:
Thanks @JDuchniewicz 

The `LTO_PLUGIN_DIR = $(HOME)/.arduino15/packages/...` doesn't look very cross platform as it does not exist on macOS.

Are those changes upstream already? If so, I'll just rebase my fork to get them.

JDuchniewicz, 2021-04-22:
I did not push the LTO_PLUGIN changes upstream as they are relevant only to this project - without including additional libs with the `ar` command (as is done normally) it does not complain. The one thing they did in the past was to replace the linking process with the g++ linker instead of gcc one.

JDuchniewicz, 2021-04-22:
I'm not sure how to do it in a future proof manner... As the versions will be changing and we have many platforms to serve. The easiest would be asking the user to provide it themselves but this might be asking for too much in some cases.